pylayers/pylayers #216: Generating trajectories using TA-OFFICE

MahiAbdelbar, 2015-01-13:
Hello,

When using the the "TA-OFFICE.ini" for generating trajectories, when the roomID, which according to my understanding is the room where the agent begins the movement, when it's equal to 4,6 or 17, the agent never leaves the room. DO you have an explanation for this ?

Thanks,

niamiot, 2015-01-13:
Hi,
Thank you for your report.

I seems you've spoted a bug in the code, I'll check this right now.

However, do you experiment this "never leave the room" behavior after several call to runsimul() ?
In this case,  maybe you should try to quit and re-launch your current python session and then re-run a simulation. 
Indeed, for some reason (seems due to a Simpy 2.x library limitation), agents positions are kept from a run to another and may interfere in the simulation.

MahiAbdelbar, 2015-01-13:
Hey, thanks for the quick response :)
So a couple of observations:
- It's actually rooms 5 and 17, the two rooms in he upper right corner and the lower right corner (what ever their ID's are, because I noticed that according to the layout in "TA-OFFICE.ini", those rooms should be 4 and 6. However, when I run the simulation, they respond to numbers 5 and 17 instead).
- The problem also exists on the other way around. So if an agent is trying to enter one of those rooms, he gets stuck on the outside of the door till the end of the simulation and doesn't get in.
- In two different scenarios, I noticed that after after agent X being stuck for a while, if another agent Y tries to enter the same room, somehow when agent Y gets really close, agent X begins to move. Eventually now agent Y gets stuck.

Anyway, thought this could help. Thanks.

niamiot, 2015-01-13:
Ok I found several issues.

One of the most important is that there's a typo in the the radius value of the agent. Due to this typo, agent are too "fat" and can't pass through narrow doors....

I'm also currently correcting other related bugs which should globally improve the mobility simulation.

I'll post a fix in couple of minutes.

closed 2015-01-14 by niamiot