pylayers/pylayers #217: Zero Division Error

MahiAbdelbar, 2015-01-13:
Another question:
I sometimes get that error while running S.runsimul()

ZeroDivisionError: float division by zero

The whole Traceback is as follows:

  File "<ipython-input-10-e96c93b86265>", line 1, in <module>
    S.runsimul()

  File "build\bdist.win-amd64\egg\pylayers\simul\simulnet.py", line 378, in runsimul
    rel_speed=float(self.sim_opt['speedratio']))

  File "C:\Anaconda\lib\site-packages\simpy-2.2-py2.7.egg\SimPy\SimulationRT.py", line 81, in simulate
    self.step()

  File "C:\Anaconda\lib\site-packages\simpy-2.2-py2.7.egg\SimPy\Simulation.py", line 525, in step
    resultTuple = proc._nextpoint.next()

  File "build\bdist.win-amd64\egg\pylayers\mobility\transit\Person.py", line 296, in move
    acceleration = self.steering_mind(self)

  File "build\bdist.win-amd64\egg\pylayers\mobility\transit\SteeringBehavior.py", line 516, in queue_steering_mind
    acceleration += behavior.calculate(boid)

  File "build\bdist.win-amd64\egg\pylayers\mobility\transit\SteeringBehavior.py", line 419, in calculate
    offset = offset.scale(radius_ij - distance)

  File "build\bdist.win-amd64\egg\pylayers\mobility\transit\Person.py", line 82, in scale
    return self.normalize() * size

  File "build\bdist.win-amd64\egg\pylayers\mobility\transit\vec3.py", line 417, in normalize
    nlen = 1.0/math.sqrt(self*self)

ZeroDivisionError: float division by zero

Can you help with that? Thanks

niamiot, 2015-01-14:
I suppose that this issue occurs when 2 agent are initialized in the same room at the same place. This bug has been resolved in the master branch.

I guess you're using the latest release and not the current developed version ? 
The last release (0.4.1) is quite outdated for the mobility modules (Our effort was focused on the RayTracing tool)

If you can , it would be more convenient for you to clone the pylayers repository. then you'll be able to get fresh update regularly.

MahiAbdelbar, 2015-01-14:
Thank you for your response.

I have cloned everything, and still I get the same error when setting two agents' roomIds to -1 at the same time, or even when actually setting two agents to the same roomId. 
Should I be getting this error still?

niamiot, 2015-01-15:
I've just take a quick look and I can't reproduce the bug. I will investigate this shortly, and I come back to you ASAP.

MahiAbdelbar, 2015-01-15:
Thank you. I'll keep investigating too and update you.

niamiot, 2015-01-16:
From my side I'm able to run a simulation with 2 agents initialized in the same room. 
Regarding the code, this should not occur because the buggy line 417 nlen = 1.0/math.sqrt(self*self) is inside a try/except which manage this Zero division

One explanation is that python  (unintentionally) still use the previously installed 'released version' of pylayers instead of the last master branch.
The best option to validate this hypothesis is to remove everything from pylayers and try a re-install from the cloned directory.

Regarding your traceback, you're using Anaconda and Windows, so you can remove everything about pylayers in this directory:
\Anaconda\Lib\site-packages
and re-install pylayers from the github cloned source (master branch)

Let me know if it helps.

MahiAbdelbar, 2015-01-16:
Yes that was really helpful. 
The cloned installation was fine as I checked the code for the try/except .
It turned out that it was the installation inside the Anaconda lib. Now It's working fine.
Thank you so much,

closed 2015-01-16 by MahiAbdelbar

niamiot, 2015-01-18:
Glad to hear you solve your problem :)