pylayers/pylayers #221: W2PTIN layout

MahiAbdelbar, 2015-02-11:
Question: what does the "W2PTIN" layout present? Why there are very few doors and most of the rooms/partitions are completely closed without any doors? There is very limited mobility. 

Thanks,

buguen, 2015-02-12:
Yes that is true this layout is unfinished. It has not been used for mobility purpose yet. 
It should evolved in a near future because there exist measurement associated to this layout.

MahiAbdelbar, 2015-02-12:
Can you tell me what it is? A residential building? An offices building? The Layout looks unique.

MahiAbdelbar, 2015-02-13:
Another question, I have added some doors around. But a specific error appears at some point:
"area error 35.142292"
and it refers to 
\build\bdist.win-amd64\egg\pylayers\gis\cycles.py(666)split()

Can you tell me what causes that error?

buguen, 2015-02-14:
Dear Mahi,

Can you send me the new version of the file you are using. Notice that it is very important that when you add new doors you add a segment which has an attribute transition=True. It is required that the room remains closed in order to be detected as a cycle in the Gs graph.
I acknowledge that it is quite tricky at that point to generate a well behaved  layout. Your attempt to try is a good way for us to improve the layout class.

Be aware, that there exist a check method which can be of help for you.
This method checks the layout integrity. 

For modifying the layout, are you using the josm plugin or are you using L.editor() ?

FYI, W2PTIN is the building office of PTIN in portugal. We have a huge dataset of hetetgeneous RAT measured at the same place and at the same time in this building during the WHERE2 project.You can find information on this
measurement campaing in the deliverable section of  the WHERE2 project. I will send you the exact deliverable url.later.

Which part of pylayers are u using most.
FYI i have worked recently on coverage ( which is quite fast) and fixed some errors. I suggest you git pull the last version of pylayers from time to time.

Thanks again for using pylayers and helping us making it a better tool 

Best regards.

Bernard

-------- Message d'origine --------

buguen, 2015-02-14:
http://www.kn-s.dlr.de/where2/documents_deliverables_bibtex.php

MahiAbdelbar, 2015-05-19:
I tried using the editor, but for some unknown reason the dialog boxes do not open up on my Mac. I actually edited the .ini file manually. Do you want to check the edited file?