pylayers/pylayers #239: Tokyo Cabinet and imposm.parser installation

kaustavkar16, 2015-08-28:
Hello Sir
     
While installing the tokyo cabinet and imposm.parser following errors occur

i> For tokyo cabinet installation it is showing:
fatal error C1083: Cannot open include file: 'tcbdb.h' : No such file or directory
error: command C:\\Users\\....\\Visual C++ for Python\\VC\\Bin\\amd64\\cl.exe failed with exit status 2

ii> For imposm.parser installation it is showing:
C:\\Users\\....\\Visual C++ for Python\\VC\\Include\\xlocale<342> warning C4530: C++ exception handler used but unwind semantics are not enabled. Specify /EHsc
...\imposm.parser\imposm\parser\pbf\osm.pb.h<9> fatal error C1083: Cannot open include file: 'Google/protobuf/stubs/Common.h No such file or directory
error: command C:\\Users\\....\\Visual C++ for Python\\VC\\Bin\\amd64\\cl.exe failed with exit status 2

protobuf and protoc are already installed.
Can you please suggest a fix for these errors.
[image]

buguen, 2015-09-01:
Could you show me the version of setup.py you are using. The module imposm is not required for pylayers to run. It just provides an interface with the open street map format in layout.py.  

comment this line in layout.py 
# import pylayers.gis.osmparser as osm

You probably have a missing library for building tokyo cabinet

kaustavkar16, 2015-09-01:
Thank you sir for the fix.

closed 2015-09-01 by kaustavkar16