pylayers/pylayers #240: Layout creation problem

kaustavkar16, 2015-09-03:
Sir
   I manually tried to create a layout of two rooms in an .ini file. The following code of the .ini file is before building the layout :- 
[image]
[image]

After this when I run L.build(), the 7 th segment gets deleted from the code automatically and the code looks like this :- 

[image]

and the output of the layout becomes

[image]

Sir can you please tell why does this thing happen.
And it would be of great help if you could suggest a solution to complete this layout without any missing segment.

Thanks

buguen, 2015-09-21:
Thank you for your post. 
Could you please paste directly the ini file of your layout in the wiki.
It would be easier for me to test what is happening here.

buguen, 2015-09-21:
It is Ok, I have found the same problem. I am fixing it

buguen, 2015-09-21:
This is fixed. The problem came from the consistency checking of the layout which was badly implemented. Get the last version and it should work.

buguen, 2015-09-21:
[info]
npoints = 8
nsegments = 9
nsubsegments = 3

[points]
-2 = (10.0, 0.0)
-8 = (10.0, 2.5)
-7 = (10.0, 7.5)
-6 = (0.0, 10.0)
-5 = (10.0, 10.0)
-4 = (20.0, 10.0)
-3 = (20.0, 0.0)
-1 = (0.0, 0.0)

[segments]
1 = {'ncycles': [1, 0], 'z': (0.0, 10.0), 'name': 'PARTITION', 'ss_name': ['METAL', 'METAL', 'METAL'], 'offset': 0, 'ss_offset': [5, 10, 20], 'transition': False, 'ss_z': [(0.0, 3), (3, 7), (7, 10)], 'norm': array([-1.,  0.,  0.]), 'connect': [-8, -7]}
2 = {'name': 'WALL', 'transition': False, 'ncycles': [], 'connect': [-8, -2], 'offset': 0, 'z': (0.0, 10.0), 'norm': array([ 1.,  0.,  0.])}
3 = {'name': 'WALL', 'transition': False, 'ncycles': [], 'connect': [-7, -5], 'offset': 0, 'z': (0.0, 10.0), 'norm': array([-1.,  0.,  0.])}
4 = {'name': 'WALL', 'transition': False, 'ncycles': [], 'connect': [-6, -1], 'offset': 0, 'z': (0.0, 10.0), 'norm': array([ 1.,  0.,  0.])}
5 = {'name': 'WALL', 'transition': False, 'ncycles': [], 'connect': [-6, -5], 'offset': 0, 'z': (0.0, 10.0), 'norm': array([ 0.,  1.,  0.])}
6 = {'name': 'WALL', 'transition': False, 'ncycles': [], 'connect': [-5, -4], 'offset': 0, 'z': (0.0, 10.0), 'norm': array([ 0.,  1.,  0.])}
7 = {'name': 'WALL', 'transition': False, 'ncycles': [], 'connect': [-4, -3], 'offset': 0, 'z': (0.0, 10.0), 'norm': array([ 1.,  0.,  0.])}
8 = {'name': 'WALL', 'transition': False, 'ncycles': [], 'connect': [-3, -2], 'offset': 0, 'z': (0.0, 10.0), 'norm': array([ 0., 1.,  0.])}
9 = {'name': 'WALL', 'transition': False, 'ncycles': [], 'connect': [-2, -1], 'offset': 0, 'z': (0.0, 10.0), 'norm': array([ 0., 1.,  0.])}

[display]
edlblsize = 20
layer = []
overlay = False
scaled = True
subseg = True
subsegnb = True
layers = ['WALL', 'PARTITION']
inverse = False
ndlblsize = 20
title =
fileoverlay =
edlabel = False
ticksoff = True
nodes = False
visu = False
edges = True
fontsize = 20
ndsize = 10
alpha = 0.5
box = (0.0, 20.0, 0.0, 10.0)
activelayer = WINDOW_GLASS
clear = False
transition = True
thin = False
ndlabel = False
layerset = ['WINDOW_GLASS', 'PLASTERBOARD_7CM', 'WALL', 'AIR', 'WINDOW', 'METALIC', 'PLASTERBOARD_14CM', 'DOOR', 'FLOOR', 'METAL', 'PARTITION', 'CONCRETE_20CM3D', 'PLASTERBOARD_10CM', 'CEIL', 'CONCRETE_6CM3D', 'CONCRETE_15CM3D', '3D_WINDOW_GLASS', 'WALLS', 'WOOD', 'CONCRETE_7CM3D', 'PILLAR', 'ABSORBENT']
ednodes = False

[files]
materials = matDB.ini
slab = slabDB.ini
furniture =

buguen, 2015-09-21:
Notice that it is useless to fill ncycles and norm which are computed from nodelista and edgelist

closed 2015-09-21 by buguen