pylayers/pylayers #249: Key error in DL.eval

kaustavkar16, 2015-12-01:
DL=DLink(L=S.L)
DL.a= tx
DL.b= rx
DL.eval(force=['sig','ray','Ct','H'],alg=2015,si_reverb=2,cutoff=2,ra_vectorized=True)

error message:

KeyError                                  Traceback (most recent call last)
<ipython-input-83-f3c74a5196d9> in <module>()
      2 DL.a= tx
      3 DL.b= rx
----> 4 DL.eval(force=['sig','ray','Ct','H'],alg=2015,si_reverb=2,cutoff=2,ra_vectorized=True)

/home/kaustav/anaconda/lib/python2.7/site-packages/pylayers-0.1-py2.7.egg/pylayers/simul/link.pyc in eval(self, **kwargs)
   1253             if kwargs['alg']==2015:
   1254                 TMP=Si.run2015(cutoff=kwargs['cutoff'],
-> 1255                         cutoffbound=kwargs['si_reverb'])
   1256             if kwargs['alg']==20152:
   1257                 TMP=Si.run2015_2(cutoff=kwargs['cutoff'],

/home/kaustav/anaconda/lib/python2.7/site-packages/pylayers-0.1-py2.7.egg/pylayers/antprop/signature.pyc in run2015(self, source, target, cutoff, cutoffbound)
   1781                     io={}
   1782                     for i in vinT:
-> 1783                         io = self.propaths2015_2(sGi,i,o,dout=io,M=M,cutoff=cutoffbound)
   1784                     di[0,0,0,o[0],o[1],o[2]] = io
   1785                     # add direct signature

/home/kaustav/anaconda/lib/python2.7/site-packages/pylayers-0.1-py2.7.egg/pylayers/antprop/signature.pyc in propaths2015_2(self, G, source, target, dout, M, Mmap, cutoff)
   1280         visited = [source]
   1281         # stack is a list of iterators
-> 1282         stack = [iter(G[source])]
   1283         # lawp = list of airwall position in visited
   1284         lawp = []

/home/kaustav/anaconda/lib/python2.7/site-packages/networkx/classes/graph.pyc in __getitem__(self, n)
    317         {1: {}}
    318         """
--> 319         return self.adj[n]
    320 
    321 

KeyError: (-32,)

Sir can you please point out reason for this error and suggest a suitable fix?

srijanrio, 2015-12-02:
It is also  happening with other layout like DLR2.  Sir can you please help us with this as we can't figure out what is causing the error.

DL=DLink(L=S.L)
DL.a= tx
DL.b= rx
DL.eval(force=['sig','ray','Ct','H'],alg=2015,si_reverb=2,cutoff=2,ra_vectorized=True)

Links save file for DLR2.ini does not exist.
Creating file. You'll see this message only once per Layout
checkh5
Start Signatures
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-15-f3c74a5196d9> in <module>()
      2 DL.a= tx
      3 DL.b= rx
----> 4 DL.eval(force=['sig','ray','Ct','H'],alg=2015,si_reverb=2,cutoff=2,ra_vectorized=True)

C:\Anaconda\lib\site-packages\pylayers-0.1-py2.7.egg\pylayers\simul\link.pyc in eval(self, **kwargs)
   1244             if kwargs['alg']==2015:
   1245                 TMP=Si.run2015(cutoff=kwargs['cutoff'],
-> 1246                         cutoffbound=kwargs['si_reverb'])
   1247                 print "algo 2015"
   1248             if kwargs['alg']==20152:

C:\Anaconda\lib\site-packages\pylayers-0.1-py2.7.egg\pylayers\antprop\signature.pyc in run2015(self, source, target, cutoff, cutoffbound)
   1790                     io={}
   1791                     for i in vinT:
-> 1792                         io = self.propaths2015_2(sGi,i,o,dout=io,M=M,cutoff=cutoffbound)
   1793                     di[0,0,0,o[0],o[1],o[2]] = io
   1794                     # add direct signature

C:\Anaconda\lib\site-packages\pylayers-0.1-py2.7.egg\pylayers\antprop\signature.pyc in propaths2015_2(self, G, source, target, dout, M, Mmap, cutoff)
   1289         visited = [source]
   1290         # stack is a list of iterators
-> 1291         stack = [iter(G[source])]
   1292         # lawp = list of airwall position in visited
   1293         lawp = []

C:\Anaconda\lib\site-packages\networkx\classes\graph.pyc in __getitem__(self, n)
    317         {1: {}}
    318         """
--> 319         return self.adj[n]
    320 
    321 

KeyError: (-18,)

niamiot, 2015-12-03:
I'm missing some elements to reproduce your error : 
- tx position
- rx position 

The following lines are working on the master branch:

>>> from pylayers.simul.link import *
>>> DL = DLink(L='DLR2.ini')
>>> DL.eval(force=['sig','ray','Ct','H'],alg=20152,si_reverb=2,cutoff=2,ra_vectorized=True)

Btw, have you cloned the github repository ( prefered option) or download a release ?

kaustavkar16, 2015-12-04:
The coordinates for DLR2 are:
tx: 30.0 4.0 1.2
rx: 14.0 2.0 1.2
Sir, as you said. we cloned the github repository on ubuntu but still exactly the same error pops up every time we run the the code for DLR2 and a layout that we had created. However this error does not appear for defstr, It would be of great help if you could suggest a fix for this.
The created layout and the tx and rx positions are as follows:

[info]
npoints = 90
nsegments = 102
nsubsegments = 34

[points]
-188 = (6.15, 4.1)
-186 = (4.65, 4.1)
-184 = (0.0, 4.1)
-182 = (1.95, 4.1)
-180 = (2.75, 4.1)
-178 = (3.15, 4.1)
-176 = (20.35, 4.1)
-174 = (8.3, 4.1)
-172 = (9.9, 4.1)
-170 = (11.1, 4.1)
-168 = (15.4, 4.1)
-166 = (16.6, 4.1)
-164 = (16.8, 4.1)
-162 = (17.0, 4.1)
-160 = (18.2, 4.1)
-158 = (22.5, 4.1)
-156 = (23.7, 4.1)
-154 = (23.9, 4.1)
-152 = (23.9, 5.6)
-150 = (23.9, 6.8)
-148 = (23.9, 7.5)
-146 = (0.0, 14.3)
-144 = (0.7, 14.3)
-142 = (2.5, 14.3)
-140 = (4.2, 14.3)
-138 = (6.0, 14.3)
-136 = (7.7, 14.3)
-134 = (9.5, 14.3)
-132 = (10.2, 14.3)
-130 = (11.3, 14.3)
-128 = (13.1, 14.3)
-126 = (14.3, 14.3)
-124 = (16.1, 14.3)
-122 = (17.2, 14.3)
-120 = (18.2, 14.3)
-118 = (20.0, 14.3)
-116 = (21.1, 14.3)
-114 = (22.9, 14.3)
-112 = (23.9, 14.3)
-110 = (24.6, 14.3)
-108 = (26.4, 14.3)
-106 = (28.1, 14.3)
-104 = (29.9, 14.3)
-102 = (31.6, 14.3)
-100 = (33.4, 14.3)
-98 = (34.1, 14.3)
-96 = (34.1, 7.5)
-94 = (34.1, 6.8)
-92 = (34.1, 5.0)
-90 = (34.1, 0.0)
-88 = (33.4, 0.0)
-86 = (31.6, 0.0)
-84 = (29.9, 0.0)
-82 = (28.1, 0.0)
-80 = (26.4, 0.0)
-78 = (24.6, 0.0)
-76 = (23.9, 0.0)
-74 = (23.025, 0.0)
[... 151 lines cut ...]
edges = True
fontsize = 20
ndsize = 10
alpha = 0.5
overlay_file = 
box = (0.0, 34.1, 0.0, 14.3)
overlay_flip = 
activelayer = WINDOW_GLASS
clear = False
transition = True
thin = False
ndlabel = False
overlay_axis = (0.0, 34.1, 0.0, 14.3)
layerset = ['WINDOW_GLASS', 'PLASTERBOARD_7CM', 'WALL', 'AIR', 'WINDOW', 'METALIC', 'PLASTERBOARD_14CM', 'DOOR', 'FLOOR', 'METAL', 'PARTITION', 'CONCRETE_20CM3D', 'PLASTERBOARD_10CM', 'CEIL', 'CONCRETE_6CM3D', 'CONCRETE_15CM3D', '3D_WINDOW_GLASS', 'WALLS', 'WOOD', 'CONCRETE_7CM3D', 'PILLAR', 'ABSORBENT']
ednodes = False

[files]
materials = matDB.ini
slab = slabDB.ini
furniture = 

Coordinates of the created layout are:

tx: 23.1 4.15 3.0
rx: 23.1 3.0 3.0

srijanrio, 2015-12-05:
@niamiot Sir the error is coming irrespective of tx rx positions. It would be very helpful if you kindly look into it.

niamiot, 2015-12-05:
I can also reproduce the bug. As a first look I cannot recommend any easy workaround, I need to investigate this.

kaustavkar16, 2015-12-05:
Thanks a lot sir.

niamiot, 2015-12-05:
I'm not sure why you're using this layout in particular, but if you can an option wild be to temporally use another one. Regards

Envoyé à partir d'Outlook mobile

srijanrio, 2015-12-06:
@niamiot  Sir the layout which @kaustavkar16  posted is the 2nd floor layout of our ECE department(txt file attached). We performed Motley-Keenan analysis on this layout. Now we wish to perform Ray tracing on the same that is when the bug occurred. So we tried it on two pre existing layouts DLR2 and defstr and the bug still persisted in DLR2.  
[ECE-Dept.txt](https://github.com/pylayers/pylayers/files/53128/ECE-Dept.txt)

niamiot, 2015-12-07:
Ok  @kaustavkar16 and @srijanrio . I think I have good news.

## short story :
I've just update the master branch, and commit 06a952f4911f5ae0eea619fd34b4c03fd10f5cd8 should solve the issue. So please git pull your repo.
 Some action points from your side to make it work : 
1. I correct few typos in the ECE-Dept.txt file you provided which prevent the build to correctly pass.so use preferably this one ( renamed into .ini): [ECE-Dept.txt] (https://github.com/pylayers/pylayers/files/54022/ECE-Dept.txt)
1bis - btw, how do you edit the file integrated/external edtior ? by hand ?  I'm interested if the found typo is related to a bug into the editor .
2. You should prevently remove any file related to ECE-Dept in pylayers-project/struc/gpickle/ and in pylayers-project/output
3. The following code should work:
>>> from pylayers.simul.link import *
>>> DL=DLink(L='ECE-Dept.ini')
>>> DL.eval()
>>> DL._show3()
[image]

## detail:

The issue comes from the fact, that cycles must describe a convex shape. In order to avoid users to make their layout convex, an automated process has been introduces few commit ago. But it was not yet used by default, due to some flaws.
I just correct the residual bug and set it as default. Now user can edit theyr layout without carrying about the build process.

srijanrio, 2015-12-07:
@niamiot  Sir, thank you very much  for the fix.
We created the basic structure of the layout in JSOM editor and added the segments and their dimensions manually in sublime text editor. Then we used the build() and save() functions to generate the ini file from the osm file.
Thanks again Sir. We are very grateful for your help.