pylayers/pylayers #250: Installation Issue

germanmendoza, 2015-12-03:
Hello there.
I'm having problems installing pylayers in Windows 8.1 and Linux Ubuntu 14.04.3. I have used Anaconda2-2.4.0 in both OSs. The error is the same in both systems too. Maybe it is something easy I'm missing, but I know nothing of Python and I'm not handy with Linux systems either. The screen shot of the error is:
[image]
Thanks in advance...

niamiot, 2015-12-03:
It seems there is an issue during the install of PIL library.
I suppose you have an anaconda distribution of python, so you should try : 
> conda install PIL
If you encountered the same issue , try an install from pip : 
> pip install PIL
Then try to reinstall pylayers.
Let me know if you manage to finish he install

germanmendoza, 2015-12-03:
Hi Nicolas,
I followed your instructions and the installation went Ok. Thanks a lot. After that, I wanted to see if pylayers was working Ok.  I tried the 4 lines sample code:
>>> from pylayers.simul.link import *
>>> DL = DLink()
>>> DL.eval()
>>> DL._show3()

I wrote them in a .py file that I run with python. Now I get the error it is shown in the following screen shot:

[image]

What should I do? Am I verifying if pylayers is Ok in the wrong way? Thanks again!

niamiot, 2015-12-04:
Could you please check if you have a .pylayers (note the '.' prefix) file created in your $HOME ( a.k.a. ~)
If so , please check that under 'source' the path correspond to the directory where you cloned the pylayers project (regarding your screenshot, it should be : ~/Downloads/pylayers-master)

germanmendoza, 2015-12-04:
Hi Nicolas.
I did what you said and now pylayers is working fine in Windows. However, in Ubuntu I got this error:
[image]
Thanks again!

niamiot, 2015-12-04:
It sounds like an issue with the basemap library installation.
make an extra : 
> conda install basemap
should solve the issue

germanmendoza, 2015-12-10:
Thanks Nicolas. Now it is working fine in Ubuntu too.

closed 2015-12-10 by germanmendoza