pylayers/pylayers #252: Assertion error in alignc() while calling DL.eval()

srijanrio, 2015-12-11:
While performing raytracing simulation we got an error while passing certain frequencies to the DL.eval() method. While this error pops up for certain frequencies , it  does not occur for others. The code  runs fine when frequencies such as 2.4(default parameter), 2-3 (fmin-fmax) ghz are passed as parameters. However when we pass frequencies such as 1,5, 6, 7 ghz etc  this error shows up. We checked it on 3 layouts ECE-Dept , DLR2 and defstr.
 SIr, can you kindly look into the bug and  tell us why this error may be occuring ?

layout : ECE-Dept
tx: ( 23.1 ,   4.15,   3.  )
rx : ( 23.1,   3. ,   3. )

Input:
fGHz2 = np.array([5])
DL=DLink(L=S.L,fGHz=fGHz2)
#DL=DLink(L=S.L)
DL.a= tx
DL.b= rx
DL.eval(force='sig','ray','Ct',3.25],alg=2015,si_reverb=2,cutoff=2,ra_vectorized=True,diffraction=True,fGHz=fGHz2)

Output:
checkh5
Start Signatures
algo 2015
Signatures'> from 8_7_2 saved
Stop signature 0.346801996231
Start Rays
Rays'> from 2_3_12 saved
Stop rays 1.26920890808
Ctilde'> from 3_12_20 saved
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-22-2ef4651fa794> in <module>()
      5 DL.a= tx
      6 DL.b= rx
----> 7 DL.eval(force=['sig','ray','Ct',3.25],alg=2015,si_reverb=2,cutoff=2,ra_vectorized=True,diffraction=True,fGHz=fGHz2)

/home/rio/anaconda/lib/python2.7/site-packages/pylayers-0.1-py2.7.egg/pylayers/simul/link.pyc in eval(self, **kwargs)
   1408             Cl=C.locbas(Tt=self.Ta, Tr=self.Tb)
   1409             #T channel
-> 1410             H = C.prop2tran(a=self.Aa,b=self.Ab,Friis=True,debug=True)
   1411             self.save(H,'H',self.dexist['H']['grpname'],force = kwargs['force'])
   1412         self.H = H

/home/rio/anaconda/lib/python2.7/site-packages/pylayers-0.1-py2.7.egg/pylayers/antprop/channel.pyc in prop2tran(self, a, b, Friis, debug)
   4138         #
   4139         #  (r x f ) (r x Nt x f )
-> 4140         t1 = self.Ctt * Fat + self.Ctp * Fap
   4141         t2 = self.Cpt * Fat + self.Cpp * Fap
   4142 

/home/rio/anaconda/lib/python2.7/site-packages/pylayers-0.1-py2.7.egg/pylayers/signal/bsignal.pyc in __mul__(self, u)
   2682 
   2683     def __mul__(self, u):
-> 2684         x,y1,y2 = self.alignc(u)
   2685         U = FUsignal(x, y1 * y2)
   2686         return(U)

/home/rio/anaconda/lib/python2.7/site-packages/pylayers-0.1-py2.7.egg/pylayers/signal/bsignal.pyc in alignc(self, u2)
   1129         xstop = min(u1_stop, u2_stop)
   1130 
-> 1131         assert(xstop>=xstart), "error the 2 signal have disjoint support"
   1132 
   1133         dx = min(dx1, dx2)

AssertionError: error the 2 signal have disjoint support

niamiot, 2015-12-12:
I suppose you're using an ominodirectilnnal antenna (check DL.Aa and DL.Ab). You may try either to change the antenna or to evaluate 2 points of frequency (e.g. np.array([4.9,5]) )
I thought that issue was fix with last update. Are you sure you running the last version on the master branch. If not try to git pull the repository and re-install.

srijanrio, 2015-12-13:
@niamiot  Yes Sir we pulled the latest version and re-installed it still the error persists for some values.
Interestingly, for tx position (23.1 4.15 3.0) it is working on all frequencies just for rx position (23.1 3.0 3.0). But when rx position is changed to any other position like (22.8 2.0 3.0) it is showing the assertion error. 
[layout: ECE-Dept]
 As you mentioned we changed the antenna to Gauss and its working for all positions. But , Gauss antenna isn't talking passed fGHz and is only working on the default fGHz value as seen from the link evaluation info:

fmin (fGHz) : 0.8
fmax (fGHz) : 5.95
fstep (fGHz) : 0.05

  We checked that  in @fGHz.setter  i.e. def fGHz(self,freq) in link.py there is an if  block for Aa and Ab type  'Omni' and not for other antennas like 'Gauss'.Sir,is there any specific reason for this ?

niamiot, 2015-12-14:
The problem you encountered is due to a known bug inAntenna class with omnidirectional antenna. 
Omni antennas are the  only antennas described for a single frequency ( the omni antenna is supposed have an omni pattern for any frequency. ) any other antenna have a specific radiation pattern for a given frequency. 
For the problem at hand, there is a match issue between the frequency range of the simulation and the single one owned by the antenna. 
This explain why omni antennas have a specific processing block in the dlink setter. 
The fix has to be done either in antenna class or in the channel.prop2tran 
I try to fix this ASAP.

srijanrio, 2015-12-14:
Oh alright. Thank you very much Sir.