pylayers/pylayers #254: Interference calculation in evsinr()

kaustavkar16, 2015-12-24:
Sir, in the evsinr method in coverage.py, does the calculation of interference power follow any particular model, if so then which one?

kaustavkar16, 2015-12-26:
Sir, in the same method what does the variable 'U' represent? Can you please state its function?

buguen, 2016-02-23:
I recognize this function is a little bit cryptic. 

Na is the number of APs. We are assuming that all AP are interfering at the same time which is the worst case, but in practice not realistic. You could change that situation  by modifying this U which is a kind of selector of the interferers. 
 
U is a tensor (1x1xNaxNa) 

ImWo is the total interfering power for each AP assuming all other AP are interferers (defined by the syructure of U). This is for orthogonal polarization.
ImWp is the same for the parallel polarization.