pylayers/pylayers #262: IndexError running example code

eokeeffe, 2016-05-23:
Heya
Was able to install everything grand. Only problem was running the example code
and also nose_tests 

from pylayers.simul.link import *
DL = DLink()
DL.eval()
DL._show3()

File "test.py", line 2, in <module>
    DL = DLink()
  File "build/bdist.linux-x86_64/egg/pylayers/simul/link.py", line 350, in __init__
  File "build/bdist.linux-x86_64/egg/pylayers/gis/layout.py", line 420, in __init__
  File "build/bdist.linux-x86_64/egg/pylayers/gis/layout.py", line 1498, in load
  File "build/bdist.linux-x86_64/egg/pylayers/gis/layout.py", line 1408, in loadini
  File "build/bdist.linux-x86_64/egg/pylayers/gis/layout.py", line 866, in g2npy
IndexError: too many indices

Any idea what could be the problem?

buguen, 2016-05-26:
Hello Evan, 

Thank you for rising the issue. It always a real pleasure to see some interest out there. 
I am not able to reproduce the problem on my paltform. 
What is exactly your platform. Try to pull request the latest version on the master branch. 
Concerning the nose test I confess that I have some cleaning work on my plate. Some test corresponds to older version and are no longer elevant. I will try to have a clean testing procedure asap but it is not case at that time. The input data format has been changed recently and this could cause some problem. Always happy to try finding an issue in order to improve the pylayers platform. 

Bernard

buguen, 2016-05-26:
Thhis is the output you should get
[image]