pylayers/pylayers #266: MultipleInstanceError

rymnikski, 2016-06-05:
This problem  it seems to be related to versions of jupyter and ipdb.

Running in interactive shell Jupyter I get this:

MultipleInstanceError: Multiple incompatible subclass instances of TerminalIPythonApp are being created.

When I traid to create a layout. L = layout(), referred to examples of Layout class.

niamiot, 2016-06-05:
Hi,

I can't reproduce with my config.
This seems to be related to this : 
http://stackoverflow.com/questions/35613249/using-ipdb-to-debug-python-code-in-one-cell-jupyter-or-ipython

Which version are you using ? the master or last release ?

rymnikski, 2016-06-05:
Hi,

I'm using the master.

rymnikski, 2016-06-05:
Hi, I solved. 

The problem was with ipbd version 0.10 that is incompatible  for now with Ipython, so I'm now using ipbd == 0.8

pip install ipdb==0.8

niamiot, 2016-06-06:
Ok nice :)
Thanks for the solution.

closed 2016-06-06 by niamiot