pylayers/pylayers #273: ERROR Loading an outdoor layout from its address

dialounke, 2016-08-29:
Hello, 

I tried to use the Layout for an outdoor from the address or coordinates but I have error and I can't understand ad fix the problem.

I obtained this following log error : 

from pylayers.gis.layout import *
import numpy as np
import seaborn as sb
import pandas as pd
import matplotlib.pyplot as plt
L = Layout('Servon sur Vilaine',dist_m=100) # or L=Layout('(60.204378,24.659540)',dist_m=300)
L.showG('s')
plt.show()`

NoOptionError                             Traceback (most recent call last)
<ipython-input-2-006a8dcc7908> in <module>()
----> 1 L = Layout('Servon sur Vilaine',dist_m=100)
      2 # Show the layout
      3 L.showG('s')
      4 plt.show()

/home/dialounke/code/pylayers/pylayers/gis/layout.pyc in __init__(self, string, _filematini, _fileslabini, _filefur, force, check, build, verbose, cartesian, dist_m)
    235                 self.importosm(latlon=string,dist_m=dist_m,cart=True)
    236             else:
--> 237                 self.importosm(address=string,dist_m=dist_m,cart=True)
    238 
    239 

/home/dialounke/code/pylayers/pylayers/gis/layout.pyc in importosm(self, **kwargs)
   1115         # get slab and materials
   1116         #
-> 1117         mat = sb.MatDB()
   1118         mat.load(self.filematini)
   1119         self.sl = sb.SlabDB()

/home/dialounke/code/pylayers/pylayers/antprop/slab.pyc in __init__(self, _fileini, dm)
    865         self.filemat = self.fileini.replace('.ini','.mat')
    866         if dm=={}:
--> 867             self.load(_fileini)
    868         else:
    869             #self.update(dm)

/home/dialounke/code/pylayers/pylayers/antprop/slab.pyc in load(self, _fileini)
   1112         for k,matname in enumerate(materials.sections()):
   1113             M = Mat(name=matname)
-> 1114             M['sigma'] = eval(materials.get(matname,'sigma'))
   1115             M['roughness'] = eval(materials.get(matname,'roughness'))
   1116             M['epr'] = eval(materials.get(matname,'epr'))

/home/dialounke/anaconda2/lib/python2.7/ConfigParser.pyc in get(self, section, option, raw, vars)
    616             value = d[option]
    617         except KeyError:
--> 618             raise NoOptionError(option, section)
    619 
    620         if raw or value is None:

NoOptionError: No option 'sigma' in section: 'dict'`

Someone can help me please?

Thanks

buguen, 2016-08-29:
Could you check in your file matDB.ini that all material have a sigma option

[ABSORBENT]
epr = (1+0j)
mur = (1+0j)
sigma = 0.0
roughness = 0.0
...

buguen, 2016-08-29:
NoOptionError: No option 'sigma' in section: 'dict'`

dialounke, 2016-08-29:
Yes, there is sigma option. Here-below, my matDB.ini file in the Pylayers_project/ini : 

[dict]
0 = ABSORBENT
1 = AIR
2 = BRICK
3 = PLASTER
4 = GLASS
5 = CONCRETE
6 = REINFORCED_CONCRETE
7 = WOOD
8 = STONE
9 = SEA_WATER
10 = PLATRE-57GHz
11 = CONCRETE-METIS-TC1
12 = WOOD-METIS-TC1
-1 = METAL

[ABSORBENT]
epr = (1+0j)
mur = (1+0j)
sigma = 0.0
roughness = 0.0
index = 0

[AIR]
epr = (1+0j)
mur = (1+0j)
sigma = 0.0
roughness = 0.0
index = 1

[BRICK]
epr = (4.09999990463+0j)
mur = (1+0j)
sigma = 0.300000011921
roughness = 0.0
index = 2

[PLASTER]
epr = (8+0j)
mur = (1+0j)
sigma = 0.038
roughness = 0.0
index = 3

[GLASS]
epr = (3.79999995232+0j)
mur = (1+0j)
sigma = 0.0
roughness = 0.0
index = 4

[CONCRETE]
epr = (5.5+0j)
mur = (1+0j)
sigma = 0.048699988079
roughness = 0.0
index = 5

[REINFORCED_CONCRETE]
epr = (8.69999980927+0j)
mur = (1+0j)
sigma = 3.0
roughness = 0.0
index = 6

[WOOD]
epr = (2.83999991417-0.019999999553j)
mur = (1+0j)
sigma = 0.0
roughness = 0.0
index = 7

[STONE]
epr = (8.69999980927+0j)
mur = (1+0j)
sigma = 3.0
roughness = 0.0
index = 8

[SEA_WATER]
epr = (81+0j)
mur = (1+0j)
sigma = 4.0
roughness = 0.0
index = 9

[PLATRE-57GHz]
epr = (3.097-0.04928j)
mur = (1+0j)
sigma = 0.0
roughness = 0.0
index = 10

[CONCRETE-METIS-TC1]
epr = (6.14-0.3j)
mur = 1
sigma = 0.0
roughness = 0.0
index = 11

[WOOD-METIS-TC1]
epr = (1.64-0.11j)
mur = 1
sigma = 0.0
roughness = 0.0
index = 12

[METAL]
epr = (-1-1j)
mur = (1+1j)
sigma = 1000.0
roughness = 0.0
index = -1

buguen, 2016-08-29:
This is the old format. 
The first section [dict] is no longer required. 
If you remove your P1 directory, it will copy the new version of the file MATDB.ini file. It contains additional materials. 

A good practice is to mentioned the reference  of the material properties as for example 

; from http://www.ic1004.org/uploads/ChannelMeasurementsModelingFor5GAbove6GHz.pdf  p 24
[CONCRETE_IC1004]
epr = (6.5+0j)
mur = (1+0j)
sigma = 0.668
roughness = 0.0

You are probably working with an old of version of P1.

dialounke, 2016-08-29:
Thanks,

It's fine now; I delete the pylayers and re-install from scratch. I generate the good Pylayers_project without error (the first had errors when I tried to generate the Pylayers_project )

closed 2016-08-29 by dialounke