pylayers/pylayers #277: This project may hard to use in windows.

wystephen, 2016-10-04:
This project use a python library named imposm which is only can be use in linux.Because the tokycabinet (http://fallabs.com/tokyocabinet/) which the imposm need for compiling is only offer a makefile version.And makefile is hard to use in windows platform.
Is there other to install the imposm package?Or,Just because I have some error when I try to install the pylayers?

wystephen, 2016-10-04:
I'd like to communicate with you about the pylayers,can I get your email?

niamiot, 2016-10-04:
Imposm is an optional install for running Pylayers. We're also currently considering remove this dependency.
You should be able to install/use Pylayers without this library