pylayers/pylayers #282: No module named mayavi.mlab

lidiaxp, 2017-09-13:
I'm an user of your simulator and for the time being didn't ecountered any trouble with it. But after updating the tool I"ve crossed with a recorring error, and couldn't find a solution anywhere on the internet, so I'm writing this issue so you can work on a solution to it. 
The error is described as follows:
 `No module named mayavi.mlab`
It occurs when we try to execute the indoor coverage of a signal.

niamiot, 2017-09-13:
Hi,

Thank for your report.
It seems that you haven't install Mayavi, which causes this error message. Anyway, the Mayavi install is theoretically optional so, something operates wrong here.
Could you please tell me more about when the error message occurs (command typed, traceback) ?

lidiaxp, 2017-09-14:
Hi,
Of course.
I used this commads to try indoor coverage code:

from pylayers.antprop.coverage import *
from matplotlib.pyplot import *
import time

C = Coverage('coverage.ini')
C.Cover()
fig=figure(figsize=(10,5))

f,a=C.show(fig=fig)
plt.show()

And appeared error: "No module named mayavi.mlab" from the first line

Attached photos.

I try install a february code of this simulator, and has been shown that:
"mayavi not installed" but the code works normally. But, with codes that
are posted after august appear this error and nothing work(nothing in
indoor coverage function). Can you help me?

meticulous-coder, 2017-10-06:
Hi All, new user here. 

@lidiaxp: did you find a solution? I'm having the same issue. Using Ubuntu 17.04. First time installing anything to do with Python on it. I installed Anaconda2 and latest version of pylayers code (i.e. from today).

I did notice the following output when running the installer_unix script:

UnsatisfiableError: The following specifications were found
to be in conflict:
- mayavi
- navigator-updater
- pywavelet

UnsatisfiableError: The following specifications were found
to be in conflict:
- basemap
- navigator-updater
- pil

PackageNotFoundError: Packages missing in current channels:
- xwpython

The rest of the installation/setup was fine. When I try running scripts I get the following output:

`br@br-VirtualBox:~/pylayers/pylayers/simul/tests$ python test_metis.py
Layout:Mayavi is not installed
Traceback (most recent call last):
  File "test_metis.py", line 1, in <module>
    from pylayers.simul.link import *
  File "build/bdist.linux-i686/egg/pylayers/simul/link.py", line 118, in <module>
  File "build/bdist.linux-i686/egg/pylayers/signal/waveform.py", line 14, in <module>
  File "build/bdist.linux-i686/egg/pylayers/measures/mesuwb.py", line 115, in <module>
  File "build/bdist.linux-i686/egg/pylayers/antprop/channel.py", line 167, in <module>
  File "build/bdist.linux-i686/egg/pylayers/antprop/antenna.py", line 151, in <module>
ImportError: No module named mayavi.mlab`

What could be causing this? Thanks.

lidiaxp, 2017-10-11:
Sorry, but they have not answered me yet. For now I'm using a code made
available in March that works normally, I believe it was some update they
did that made the multiwall model stop working. About your problem, it is
different from mine, had never seen, you run into it after running which
script?

niamiot, 2017-10-11:
hi,

Sorry for this late answer, 
Mayavi install and its compatibility (epsecially with matplolib & basemab) is clearly not obvious. 
We try to simplify as much the install process, but it s quite hard to ensure everything to be ok on amm plateforms and OS version.

I just make a (dirty) workaround which bypass the mayavi import in the antenna module, it should resolve your import issues. This comes with the last commit a42b70fa15121b275fc09abd1aac28f23e9c6cb7

Hope it helps

meticulous-coder, 2017-10-11:
Hi niamiot,

Thank you for the reply! I will try it out and report back.

meticulous-coder, 2017-10-11:
It seems to be working now. Thanks! I tested the same script and got the following output (no errors but I'm not sure if this is the correct output):

`br@br-VirtualBox:~/pylayers/pylayers/simul/tests$ python test_metis.py
Layout:Mayavi is not installed
mayavi not installed
Layout:Mayavi is not installed
Warning : imposm seems not to be installed
Layout:Mayavi is not installed
Warning Unable to read graph Gr
Warning Unable to read graph Gw
Layout Graph loaded
Signatures: 100.0it [01:25,  5.49it/s] `

I then tested this code:

`from pylayers.simul.link import *
DL = DLink()
DL.eval()
DL._show3()`

and got the following output:

`br@br-VirtualBox:~/Desktop$ python mypy.py
Layout:Mayavi is not installed
mayavi not installed
Layout:Mayavi is not installed
Warning : imposm seems not to be installed
Layout:Mayavi is not installed
Traceback (most recent call last):
  File "mypy.py", line 5, in <module>
    DL.eval()
  File "build/bdist.linux-i686/egg/pylayers/simul/link.py", line 1632, in eval
  File "build/bdist.linux-i686/egg/pylayers/simul/link.py", line 527, in ca
AttributeError: 'DLink' object has no attribute '_ca'`

Is there perhaps something else I must change to get Pylayers to work? Thank you.

niamiot, 2017-10-12:
Hi,
The Warnings from the first function call are quite normal. You may ignore them.

About the second error , you're right, something change in the call of a Dlink object, it now require to pass a Layout file.
the correct code is :

`from pylayers.simul.link import *`
`DL = DLink(L='defstr.lay') # where defstr.lay can be replaced by the layout of your choice`
`DL.eval()`
`DL._show3() # if mayavi is installed`

I'll also correct the code from the main page of the github repo.