pylayers/pylayers #283: ImportError: No module named simulem

ShaneFarrell, 2017-09-21:
I get this error whenever I import the pylayers.antprop.coverage module:
`>>> from pylayers.antprop import coverage
Warning : imposm seems not to be installed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "build\bdist.win-amd64\egg\pylayers\antprop\coverage.py", line 22, in <module>
  File "build\bdist.win-amd64\egg\pylayers\antprop\loss.py", line 39, in <module>
ImportError: No module named simulem`

And if I try to run the ex_rays.py example in antprop (with python) I get this error:
`Warning : imposm seems not to be installed
Traceback (most recent call last):
  File "ex_rays.py", line 6, in <module>
    from pylayers.simul.simulem import *
ImportError: No module named simulem`

There does not seem to be a file called `pylayers/simul/simulem.py`, so I'm guessing there's some dynamic generation going.

Also, my objective is to create simple 2D / 3D WiFi propagation simulations including walls and to get the predicted RSSI at various points. Do I need any optional libraries for that?

niamiot, 2017-09-22:
Hi,

simulem was a (very) old module which should have been completely removed from now. I'm a bit confused why it still appears in the code. 
Anyway, you can now pull the repo and your import problem should be solved.

About your coverage simulation, the module [coverage](http://pylayers.github.io/pylayers/modules/pylayers.antprop.coverage.html) ( included in the pylayers package) should help you. You may also take a look on the [Indoor Radio Coverage with Motley Keenan](http://pylayers.github.io/pylayers/auto_examples/plot_coverage.html#sphx-glr-auto-examples-plot-coverage-py) example from the website.

The configuration file _coverage.ini_ is located into your project directory ($HOME/Pylayers_Project/ini)

ShaneFarrell, 2017-09-22:
Yep, cool. That fixed it after I re-ran setup.py.

When trying to create a Coverage object with the default config I got "Key Error: 'phideg'". I resolved this by adding in the values for phideg from the tutorial into the coverage ini file.

It is now working for me in ipytohn notebook. Though for another error it seems to break when there is only one AP (and I only want to use one) with this error:
`
Warning Unable to read graph Gs
Warning Unable to read graph Gt
Warning Unable to read graph Gv
Warning Unable to read graph Gi
Warning Unable to read graph Gr
Warning Unable to read graph Gw
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-17-e26925b9c369> in <module>()
      7 C.tx = np.array((39,1))
      8 start = time.time()
----> 9 C.cover()
     10 finish = time.time()
     11 C.show(figsize=(16,9))

C:\Users\Boop\AppData\Local\Continuum\Anaconda2\lib\site-packages\pylayers-0.5-py2.7.egg\pylayers\antprop\coverage.pyc in cover(self, sinr, snr, best)
    446 
    447         if snr:
--> 448             self.evsnr()
    449         if sinr:
    450             self.evsinr()

C:\Users\Boop\AppData\Local\Continuum\Anaconda2\lib\site-packages\pylayers-0.5-py2.7.egg\pylayers\antprop\coverage.pyc in evsnr(self)
    456         """
    457 
--> 458         NmW = 10**(self.pndbm/10.)[np.newaxis,:]
    459 
    460         self.snro = self.CmWo/NmW

IndexError: invalid index to scalar variable.
`

Thank you very much for the help.

niamiot, 2017-09-22:
ok, this should be fixed with the last commit. 
Could you try & give a feedback please?

ShaneFarrell, 2017-09-25:
Thanks, it's getting closer to being fully functional.

Now when I run it with a single AP I get a different error. Though after receiving this error and running it again with two APs the output is two diagrams, the first one being the single AP diagram rendered correctly and the second being the two APs diagram. Thus I would assume the diagram is being generated correctly but there is an issue in the code telling it to render when there is only one AP.

Error:
```
Warning Unable to read graph Gs
Warning Unable to read graph Gt
Warning Unable to read graph Gv
Warning Unable to read graph Gi
Warning Unable to read graph Gr
Warning Unable to read graph Gw
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-7-e26925b9c369> in <module>()
      9 C.cover()
     10 finish = time.time()
---> 11 C.show(figsize=(16,9))
     12 print 'Tx position: ',C.tx
     13 print 'Coverage in %1.2f seconds' % (finish-start)

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\pylayers-0.5-py2.7.egg\pylayers\antprop\coverage.pyc in show(self, **kwargs)
   1053                         ax.contour(np.sum(self.bestsvo,axis=2)[f,:].reshape(self.nx,self.ny).T,extent=(l,r,b,t),linestyles='dotted')
   1054                     if polar=='p':
-> 1055                         ax.contour(np.sum(self.bestsvp,axis=2)[f,:].reshape(self.nx,self.ny).T,extent=(l,r,b,t),linestyles='dotted')
   1056 
   1057         # display access points

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\matplotlib\__init__.pyc in inner(ax, *args, **kwargs)
   1895                     warnings.warn(msg % (label_namer, func.__name__),
   1896                                   RuntimeWarning, stacklevel=2)
-> 1897             return func(ax, *args, **kwargs)
   1898         pre_doc = inner.__doc__
   1899         if pre_doc is None:

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\matplotlib\axes\_axes.pyc in contour(self, *args, **kwargs)
   5823             self.cla()
   5824         kwargs['filled'] = False
-> 5825         contours = mcontour.QuadContourSet(self, *args, **kwargs)
   5826         self.autoscale_view()
   5827         return contours

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\matplotlib\contour.pyc in __init__(self, ax, *args, **kwargs)
    863 
    864         self._process_args(*args, **kwargs)
--> 865         self._process_levels()
    866 
    867         if self.colors is not None:

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\matplotlib\contour.pyc in _process_levels(self)
   1197         # The following attributes are no longer needed, and
   1198         # should be deprecated and removed to reduce confusion.
-> 1199         self.vmin = np.amin(self.levels)
   1200         self.vmax = np.amax(self.levels)
   1201 

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\numpy\core\fromnumeric.pyc in amin(a, axis, out, keepdims)
   2370 
   2371     return _methods._amin(a, axis=axis,
-> 2372                           out=out, **kwargs)
   2373 
   2374 

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\numpy\core\_methods.pyc in _amin(a, axis, out, keepdims)
     27 
     28 def _amin(a, axis=None, out=None, keepdims=False):
---> 29     return umr_minimum(a, axis, None, out, keepdims)
     30 
     31 def _sum(a, axis=None, dtype=None, out=None, keepdims=False):

ValueError: zero-size array to reduction operation minimum which has no identity
```

niamiot, 2017-09-25:
I cannot reproduce your error, could you please provide your ini configuration file ?

ShaneFarrell, 2017-09-25:
coverage ini file:
```
[grid]
nx = 100
ny = 50
boundary = [20,0,30,20]
mode = full ; file, zone , full 
file = 'points.ini'

[layout]
filename = TA-Office.lay ; 0 40 0 15
;filename = W2PTIN.ini
;filename = Lstruc.str

[ap]
0 = {'name':'room1','wstd':'ieee80211b','p':(1,12,1.2),'PtdBm':0,'chan':[11],'on':True,'phideg':0} 

[rx]
temperaturek = 300
noisefactordb = 0 

[show]
show = True

```

code:
```
from pylayers.antprop.coverage import *
import matplotlib.pyplot as plt
import time
import pdb
C = Coverage()
C.L._filename
C.tx = np.array((39,1))
start = time.time()
C.cover()
finish = time.time()
C.show(figsize=(16,9))
print 'Tx position: ',C.tx 
print 'Coverage in %1.2f seconds' % (finish-start)
plt.show()
```

Second AP ini:
`1 = {'name':'room2','wstd':'ieee80211b','p':(10,2,1.2),'PtdBm':0,'chan':[11],'on':True,'phideg':90}`

ShaneFarrell, 2017-09-25:
Huh, actually the error has changed. Though I've found that if I actually just catch the error and pass everything works fine. I'm ok going with that, so unless you want to fix it for its own sake there's no need.

A few questions though while I'm talking to you: What method / attribute would I use from the coverage object to get the grid of RSSI values without plotting them? Also is the best way of dynamically placing walls to edit the layout file and then load it or is there an in code way of creating a layout? Basically as part of a larger algorithm I want to place walls and an AP in algorithmically determined positions and then get the RSSI at every point on a grid.

Thanks again for all the help! :)

Here's what the error changed to:
```
Warning Unable to read graph Gs
Warning Unable to read graph Gt
Warning Unable to read graph Gv
Warning Unable to read graph Gi
Warning Unable to read graph Gr
Warning Unable to read graph Gw
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-14-f78c13611b8a> in <module>()
     10 finish = time.time()
     11 # try:
---> 12 C.show(figsize=(16,9))
     13 # except ValueError:
     14 #     pass

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\pylayers-0.5-py2.7.egg\pylayers\antprop\coverage.pyc in show(self, **kwargs)
   1053                         ax.contour(np.sum(self.bestsvo,axis=2)[f,:].reshape(self.nx,self.ny).T,extent=(l,r,b,t),linestyles='dotted')
   1054                     if polar=='p':
-> 1055                         ax.contour(np.sum(self.bestsvp,axis=2)[f,:].reshape(self.nx,self.ny).T,extent=(l,r,b,t),linestyles='dotted')
   1056 
   1057         # display access points

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\matplotlib\__init__.pyc in inner(ax, *args, **kwargs)
   1895                     warnings.warn(msg % (label_namer, func.__name__),
   1896                                   RuntimeWarning, stacklevel=2)
-> 1897             return func(ax, *args, **kwargs)
   1898         pre_doc = inner.__doc__
   1899         if pre_doc is None:

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\matplotlib\axes\_axes.pyc in contour(self, *args, **kwargs)
   5823             self.cla()
   5824         kwargs['filled'] = False
-> 5825         contours = mcontour.QuadContourSet(self, *args, **kwargs)
   5826         self.autoscale_view()
   5827         return contours

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\matplotlib\contour.pyc in __init__(self, ax, *args, **kwargs)
    863 
    864         self._process_args(*args, **kwargs)
--> 865         self._process_levels()
    866 
    867         if self.colors is not None:

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\matplotlib\contour.pyc in _process_levels(self)
   1197         # The following attributes are no longer needed, and
   1198         # should be deprecated and removed to reduce confusion.
-> 1199         self.vmin = np.amin(self.levels)
   1200         self.vmax = np.amax(self.levels)
   1201 

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\numpy\core\fromnumeric.pyc in amin(a, axis, out, keepdims)
   2370 
   2371     return _methods._amin(a, axis=axis,
-> 2372                           out=out, **kwargs)
   2373 
   2374 

C:\Users\C12722625\AppData\Local\Continuum\Anaconda2\lib\site-packages\numpy\core\_methods.pyc in _amin(a, axis, out, keepdims)
     27 
     28 def _amin(a, axis=None, out=None, keepdims=False):
---> 29     return umr_minimum(a, axis, None, out, keepdims)
     30 
     31 def _sum(a, axis=None, dtype=None, out=None, keepdims=False):

ValueError: zero-size array to reduction operation minimum which has no identity
```

niamiot, 2017-09-26:
Ok if you are able to simulate it's good for you. I'll try to take a look on the error because I'm always affraid this is the tree hiding the forest of bug. 

About your questions :

**1. useful coverage attributes**

+ self.freespace : Loss @ fGHz   PL()  PathLoss  $ ( which is just theoretical losses without taking into account the layout walls. If you want the losses including the walls and the layout you have to multiply this freespace values with the wall losses Lwo or Lwp in orthogonal or parallel polaroization respectively e.g :self.freespace*self.Lwo)
+ self.CmWo  Received Power coverage in mW polar o
+ self.CmWp  Received Power coverage in mW polar p

after run self.snr()
+ self.snro : SNR polar o (H)
+ self.snrp : SNR polar p (H)

after run self.sinr()
+ self.sinro : SINR polar o (H)
+ self.sinrp : SINR polar p (H)

after run self.evbestsv()
+ self.bestsvo : Best server polar o
+ self.bestsvp : Best server polar p

Those objects are 3D numpy arrays which depends on :
+ self.nf : the number of frequencies
+ self.nx : number of grid points in x axis
+ self.ny : number of grid points in y axis
+ self.na : number of access points
Their shape are (nf, nx*ny,na).

**To sum up** , if you want to obtain the Received Power coverage in mW in polar p, at the last frequency point for the first access point:
self.CmWp[-1,:,0].reshape(self.nx,self.ny)

**2. Layout editing**

At this point, the easiest way to edit a layout is probably to use jOSM: an editor for open street map. PyLayers ( with the Layout class) is able to import .osm file format and convert it into the PyLayers' format (.lay).
We are currently working to a minimal editor to make things more straightforward for editing simple layout and update materials. This should be release in beta quite soon.

ShaneFarrell, 2017-09-28:
Ok cool, I am now dynamically generating the coverage ini and a layout file with my algorithm and then running the coverage. It's all working nicely in 2D. How would I make it 3D? I tried changing the third value in the AP's position (which was 1.2) to a different value expecting the signal to get worse since I was moving the AP upwards but nothing happened. Even setting the value to 1000 does nothing.

I see that segments have z with two values which I assume are the bottom and top of the wall.

How would I go about simulating a building with multiple floors and getting a 3D Received Power map?

yiwenCao92, 2017-09-29:
How did you solve the problem'Key Error:phideg'?I tried to add"ant:gauss "and "phideg"in the coverage.ini,but the problem still existed.Thanks.

ShaneFarrell, 2017-09-29:
I posted the ini file I'm using up above, you should try using that and see if it works for you. I think that's the only change I made. Also remember to run setup.py again when you pull an update.

niamiot, 2017-09-29:
@yiwenCao92 
As @ShaneFarrell proposed, you have to update your coverage.ini in your PyLayers_project directory, and add the 'phideg' key to the AP dict. ( 'phideg' stands for the azimuth of the antenna in degree ). 

One thing to understand is : When you install PyLayers for the first time, data are copied from the data directory of the pylayers source to your project directory. The 2 directories path can be found into ".pylayers" file (hidden file) at the root of your HOME directory.
**All the commit only update  the pylayers data directory and not your project directory**. So e.g. any changes on '.ini' files by commit must be **manually** copied to your how project directory.

niamiot, 2017-09-29:
@ShaneFarrell 
Very happy you manage to run your algorithm.
I'm a bit less happy with the problem you mentioned.

First of all I have to precise that **multiple floors are not managed for the time being**. This is mostly  layout description limitation which only accepts 1 floor and 1 ceil. So transmissions through multiple floors and ceil are not supported right now. We clearly envisage to enable this (very requested) feature.

However, the elevation of the AP should modify the coverage because the code clearly take into account the 3rd dimension. I'll check this from my side and I come back to you ASAP.

Otherwise, to answer to an oldest question about the layout edition, the Layout Editor is currelntly in development ( alpha version) . You can see a sneak peak here : https://www.youtube.com/watch?v=g_aZMPE6wqE&t

ShaneFarrell, 2017-09-29:
Whoops, I did not mean to close this.
Ok, that's a pity about the 3D. Is the raycasting done in 3D though? Is the restriction just that there can only be one floor and ceiling? Because in that case I could presumably simulate an AP on the floor above by placing the AP above the ceiling. That should be fine because it's uncommon for signals to make it through two floors anyway. And since I can specify the z value of the top and bottom of a wall I should even be able to place walls on the floor above. Is there a way to specify the z value of the floor and ceiling? If there is I could move the entire layout up and down to simulate scanning at different heights.

Also, what do o and p polar mean? They clearly have different results but I can't figure out what they mean.

yiwenCao92, 2017-09-30:
@niamiot @ShaneFarrell Thank you for your help.You two are really boss.Actually i installed pylayers in win8,not in Ubuntu.I followed @niamiot 's direction and found the directory in win.Now I can run the demo.Thanks again!

niamiot, 2017-10-02:
@yiwenCao92 
This is quite the same type of error here. 
The files SlabDB.ini & MatDB.ini are corrupted in your PyLayers_project/ini directory.
Replace them with those from the <PyLayers_sources_directory>/data/ini.
This will solve the isssue

niamiot, 2017-10-02:
@ShaneFarrell 
There was clearly an issue with the height of access points. For a weird reason, the elevation was disable in the code wheras all the angle computation was done in 3D. Anyway this has been solved in the last commit.
As well, the transmission through ceil and floor wasn't taken into account, but now they are.
You can now set an AP upper that the ceil height and this will be counted in the losses.

This commit has also add the possibility to ad dthe height where the cover grid is evaluated. To this end, you can add a zgrid = <a height value in meter here> value in the [grid] section of the coverage.ini file. ( Use coverage.ini file from PyLayers' data directory as an example).

It worth to mention that the CEIL and FLOOR slabs used in computation are described in the .lay file in section [slabs]. By default , the used slabs are REINFORCED_CONCRETE, which means quite no energy pass through and then levels are very very low. You may try to change the slabs to see the difference.

> what do o and p polar mean?
'o' and 'p' stand for orthogonal and parallel polarization repectively.
Orthogonal polarization means that the E field varies in a plan orthogonal to the floor/ceil plan (a.k.a in a plan parallel to the walls)
Parallel polarization means that the E field varies in a plan orthogonal to the walls plan (a.k.a in a plan parallel to the ceil/floor).
e.g : Supposing a Wifi Antenna on an usb dongle which is pointing the ceil, you should consider the orthogonal polarization simulation results in this situation.

ShaneFarrell, 2017-10-25:
I now have the coverage rendering in 3D and everything looks perfect except that there doesn't seem to be a ceiling. I have checked up to 25 meters upwards and at no point is there a sharp signal drop. This is with the TA-office layout so it's not me messing up the slabs or anything. The walls extend up 3 meters and then stop. After that the signal just spreads freely. The floor exists and does indeed knock off a lot of signal strength. Interestingly 25 meters up the signal strength is still much stronger than below the floor.

Also is there a way to set the ceiling height?

I could actually simulate everything I need to with just a floor by translating the walls and AP below the floor for rendering what's above the ceiling. Though I'm guessing this is a bug.

Thanks again for all your help. :)

AmanAmir, 2019-05-21:
Hello 
I am new to python and just started using pylayers. I am trying to run the simulation for ray tracing transmission channel example given on the pylayers documentation and get the same error
 from pylayers.simul.simulem import *
ModuleNotFoundError: No module named 'pylayers.simul.simulem'
Any help?