pylayers/pylayers #285: How to save data created by Pylayers?

yiwenCao92, 2017-10-09:
In "Multiwall Motley-Keenan Simulation",we can use "f,a = C.show(fig=fig" to get a figure about the RSSI strength of each location in the room.How can I save the data in the form such as [position x,position y,RSSI value]?Is there a function about saving data?I want to get this kind of data for fingerprinting algorithm.Thanks.

niamiot, 2017-10-09:
Hi
Actually there’s no save method in the coverage module. 
However, all data of interest can be found here :
 1. useful coverage attributes

self.freespace : Loss @ fGHz PL() PathLoss $ ( which is just theoretical losses without taking into account the layout walls. If you want the losses including the walls and the layout you have to multiply this freespace values with the wall losses Lwo or Lwp in orthogonal or parallel polaroization respectively e.g :self.freespace*self.Lwo)
self.CmWo Received Power coverage in mW polar o
self.CmWp Received Power coverage in mW polar p
after run self.snr()

self.snro : SNR polar o (H)
self.snrp : SNR polar p (H)
after run self.sinr()

self.sinro : SINR polar o (H)
self.sinrp : SINR polar p (H)
after run self.evbestsv()

self.bestsvo : Best server polar o
self.bestsvp : Best server polar p
Those objects are 3D numpy arrays which depends on :

self.nf : the number of frequencies
self.nx : number of grid points in x axis
self.ny : number of grid points in y axis
self.na : number of access points
Their shape are (nf, nx*ny,na).
To sum up , if you want to obtain the Received Power coverage in mW in polar p, at the last frequency point for the first access point:
self.CmWp[-1,:,0].reshape(self.nx,self.ny)

yiwenCao92, 2017-10-14:
Thank you for your help.I got the data in the form I want. For "self.CmWp[-1,:,0].reshape(self.nx,self.ny)"-1 means the last frequncy point(In my applicaiton I just let it be around 2.4GHz),and 0 means the ap number.what about the the second parameter?

niamiot, 2017-10-14:
The second parameters is the data for each point (x,y) of the grid stacked on one column. You have to reshape this column to get the data compliant with the grid format.