pylayers/pylayers #290: Problems with the simple Ray Tracing Simulation

Dsemme92, 2017-12-05:
Hi,
I am working on a project, where we want to use pylayers. Nevertheless I have really problems to install it successfully. We had to made some changes to run the simple ray tracing simulation, as shown in the documentation. But it doesn't look like the result in the documentation.

[image]

Could somebody who installed it successfully provide a list with the packages that are installed? That would help us a lot. At the moment we are working with Ubuntu 14.04 with Python 2.7 and followed the instructions on the install page.

Thanks!

Dsemme92, 2017-12-18:
So, I figured out how to replace the sender and receiver, but still the rays are not displayed in the figure.
[image]

[image]

We had to made some changes in the rays.py to make it work, as you can see in the code. Has anybody an idea what could be wrong?

Thank you!

niamiot, 2017-12-19:
Hi, sorry for that late answer.
your output is quite strange. It seems that the simulation is not yet finished.

At this point I can't decide if this s a simulation or a visualization issue.

Could you print the return of DL.R ( the repr of rays)?

Dsemme92, 2017-12-19:
Hi, no problem! Thanks for your answer.

To make it more easy I use now the simple 4 line Ray Simulation.

[image]

If I run this file, I get the following output.

[image]

My guess is that something is wrong with my environment or the installation.

Thank you for your help.

niamiot, 2017-12-19:
Ok thanks. 

Could you just execute the code ( until line 10 ) and type DL.R ( without parenthesis) in your terminal and show me the output.

Dsemme92, 2017-12-19:
Here is the Output.

[image]

niamiot, 2017-12-20:
Ok. So this is a RT problem, not a visualization one.

I also face this problem, this a clearly a bug. 
I'll investigate this, meanwhile, I can give you a workaround:

```
from pylayers.simul.link import *
L=Layout('defstr.lay')
rx = np.array([8,4,1.6])
tx = np.array([2,3,1.5])
L.build()
L.dumpw() # <= This is the trick
DL=DLink(L=L,a=tx,b=rx)
DL.eval(force=True)
```

L.dumpw() saves the graphs obtained from L.build(). It avoids to do again and again this graph building step (which can be quite long for complex layouts). For some reasons, if graphs are not saved ( which is normally the default behaviour), DL is not able to get them directly from the Layout object.... This will be fixed.

Dsemme92, 2017-12-31:
Hi,
sorry for the late response. 

I tried your workaround in my virtual machine, but the visualization is still without any rays.  Here is my console output.

[image]
[image]

As you can see, I did this step by step. If I execute this code as a .py-file, i don't get the mayavi error.

Thank you for your support!

niamiot, 2018-01-08:
I don't understand the output of the DL.eval call.
Here my output by using the code I provided :

> Warning Unable to read graph Gr
> Warning Unable to read graph Gw
> Layout Graph loaded
> Signatures: 100.0it [00:00, 1575.72it/s] 

The most strange part is : the returned tuple contains 2 Ray objects. The 1st one shows there are 24 rays ( which is compliant with my execution).

If you cloned the repo maybe you should try to pull the last update ot the master branch.
You may also try to delete the pylayers_project directory at the root of your home directory before re-run the code.

niamiot, 2018-03-07:
Causes highlighted in  #290

closed 2018-03-07 by niamiot