pylayers/pylayers #296: Installing problem

hadi2f244, 2018-02-27:
Some packages like mayavi using pyqt4 and there is no update for them recently. So they conflict with packages like seaborn and PIL. 
Package versions in installer_unix file must be mentioned.

niamiot, 2018-03-07:
Thank you for this remark.
We clearly have a work to do on the install procedure which is (still) quite complicated for most users.
Any comment as you just done with this issue is very welcome.
we'll try to update this ASAP

hadi2f244, 2018-03-07:
Please mention your current installed package versions in installer_unix.

kiraseya, 2018-03-08:
Same issues with install, it wasn't working and then when I tried to install the necessary versions of modules the example code seg faulted.

200729, 2018-10-15:
The installation took about 6 hours. Some packages was missing and needed to install manually. The 'Ray-Tracing simulation in 4 lines' does not look like showed on website, it is more let's say 2D not 3D, so I have no confirmation if the environment works correctly. Also 'auto_examples' does not work properly, only some of them runs, some of them is not running. If examples are not running at all, the pylayers is not installed correctly or the examples are broken?

200729, 2018-10-15:
Warning : imposm seems not to be installed

200729, 2018-10-15:
[image]
>>> !date
  File "<stdin>", line 1
    !date
    ^
SyntaxError: invalid syntax
>>>

200729, 2018-10-15:
What is line '%matplotlib inline'? it is not recognised by python, should '%' introduce a comment or something, not a '#'?

200729, 2018-10-15:
[image]
Error 404

200729, 2018-10-15:
Traceback (most recent call last):
  File "./test1.py", line 2, in <module>
    from pylayers.simul.simulem import *
ModuleNotFoundError: No module named 'pylayers.simul.simulem'

maxstrobel, 2018-11-02:
@200729: Could you  manage to install PyLayers? Seems, that I am facing the same issues.

BTW: Do you have a working environment.yml with package versions, that work?

200729, 2018-11-02:
> @200729: Could you manage to install PyLayers? Seems, that I am facing the same issues.
> 
> BTW: Do you have a working environment.yml with package versions, that work?

Hi @maxstrobel I'm not sure if my pylayers is correctly installed. The example 'Ray-Tracing simulation in 4 lines' does not look as it is shown in the picture as I wrote before. Hmm how to check the installation?

Where can I find the file environment.yml, what is it for?

When some package was missing I went to ~/anaconda3/bin and use ./anaconda install or ./pip install commands from there, it helped. Also I discovered that code shown in documentation http://pylayers.github.io/pylayers/notebook/1-GIS/Layout.html should be run by ipython not python. It is also important if you want to run code from file, the filename has to be ~.ipy not ~.py.

maxstrobel, 2018-11-02:
Hi @200729, thanks for your fast reply :)
The example in 4 lines is in my installation also very two dimensional:
[image]

However, if you play around a little bit with the parameters and set the cutoff to 10, the rays become 3d:
```
>>> from pylayers.simul.link import *
>>> DL = DLink(L='defstr.lay')
>>> DL.eval(cutoff=10)
>>> DL._show3()
```
[image]

I tried also other examples from the documentation, like

- http://pylayers.github.io/pylayers/auto_examples/plot_coverage.html
- http://pylayers.github.io/pylayers/auto_examples/plot_exLuebbers.html

and the results match exactly, so I think my installation is ok.

Regarding the _environment.yml_, that's a file to share an environment with all the installed packages: 
- https://conda.io/docs/user-guide/tasks/manage-environments.html#creating-an-environment-from-an-environment-yml-file
- https://conda.io/docs/user-guide/tasks/manage-environments.html#sharing-an-environment
- https://github.com/pylayers/pylayers/issues/315#issuecomment-435374741

200729, 2018-11-02:
Thank you @maxstrobel  example in 4 lines looks little better now.

In my environment examples from documentation you mentioned are not working:
plot_coverage
UnboundLocalError: local variable 'lonmin' referenced before assignment
plot_exLuebbers
TypeError: 'dict_values' object does not support indexing

So i suppose my environment it not working at all.
Could you please share your environment.yml ?

maxstrobel, 2018-11-02:
https://github.com/pylayers/pylayers/files/2542502/environment.txt is the file. I renamed it to .txt because .yml is not allowed as an upload.

200729, 2018-11-03:
@maxstrobel  I tried your environment file and examples worked, the results were not exactly the same, but it generally worked, thanks again!
Warning : imposm seems not to be installed
Warning Unable to read graph Gs
Warning Unable to read graph Gt
Warning Unable to read graph Gv
Warning Unable to read graph Gi
Warning Unable to read graph Gr
Warning Unable to read graph Gw
[image]

Warning : imposm seems not to be installed
Warning Unable to read graph Gr
Warning Unable to read graph Gw
[image]

I think that the major difference is that I use python 3.6 not 2.7 in my environment and probably that's why I have errors 'UnboundLocalError: local variable 'lonmin' referenced before assignment' and 'TypeError: 'dict_values' object does not support indexing'. Maybe this examples will not work with python 3.6.

Here is my environment
[environmentBase.txt](https://github.com/pylayers/pylayers/files/2544885/environmentBase.txt)

200729, 2018-11-03:
That is true that installation is very painfull and even if you consider pass that challenge the pylayers is not working at all. @niamiot would it be possible to deliver virtual machine witch pylayers version which is considered stable or maybe use docker?

niamiot, 2018-11-03:
Hi
I’m very sorry for all troubles you face during the installation. 
Your conclusions are  though, but justified. 
The docker option is currently actively considered. 
From my side, I can’t be involved in the project much as I will, and as I used to be in the past. 
I’ll try ( with the other maintainers ) to propose a solution to increase the user experience. 
By the way I’d like to thank @maxstrobel for all the patience, information and support in this issue.

200729, 2018-11-04:
@maxstrobel if you run run-tests script do they pass?

maxstrobel, 2018-11-05:
@niamiot: Anyway, thanks for the great project and your work!

@200729: I will run the tests later and report the outcome.

maxstrobel, 2018-11-05:
@200729: The tests are not working in my environment. I had to install `scikit-learn` in order to run the tests. It does not exceed `Start optimization`:
```
~/pylayers$ ./run-tests 
Start optimization
^C                                                                                        
----------------------------------------------------------------------                                     
Ran 0 tests in 1961.600s                                                                                   
                                                                                                                     
OK  
```

FizzyP-Actall, 2019-01-03:
I'd love to use pylayers but I can't quite get the install to work.  TL;DR, after install `~/pylayers_project/` is empty.

Longer explanation: I'm very close.  On Ubuntu, I've carefully gone through the install directions here: https://github.com/pylayers/pylayers/blob/master/INSTALL.md.  There were some hitches here and there but by manually installing some of the python packages I managed to get through the process.  When I try to `>>> from pylayers.simul.link import *` I get
```
PermissionError: [Errno 13] Permission denied: '/home/user/pylayers_project/project.conf'
```
which I can only assume is because the `pylayers_project` folder is empty and there is no `.conf` file..  What script do I need to run to populate the project/config folder?

Aside: I also spent a bunch of time attempting to install pylayers on windows without much success.  However, `installer.bat` did populate the `pylayers_project` folder when I ran it.  I didn't see any similar behavior from `installer_unix`.  Like I said, this library looks really cool and I'd love to use but after two days of trying I still can't get it installed properly in either OS.  What script do I need to run to populate the project/config folder?

200729, 2019-01-04:
Hi,Installation is so painfull that I would say just don't install it :PI prepared virtual machine Ubuntu 18.04 with pylayers installation I consider the most stable. I'll check if your trouble occurs also there, if not I can share this vm with you if you want.Br,Sebastian

rafaelalexandrou, 2019-06-10:
> Hi,Installation is so painfull that I would say just don't install it :PI prepared virtual machine Ubuntu 18.04 with pylayers installation I consider the most stable. I'll check if your trouble occurs also there, if not I can share this vm with you if you want.Br,Sebastian

Dear Sebastian, I am having the same issues.  would you mind sharing your VM with me as well?

200729, 2019-06-11:
Hi,Yes, no problem. Could you please send your email address?

rafaelalexandrou, 2019-06-12:
r.alexandrou@hotmail.com
Thanks in advance

lzyang1995, 2019-10-04:
> Hi,Installation is so painfull that I would say just don't install it :PI prepared virtual machine Ubuntu 18.04 with pylayers installation I consider the most stable. I'll check if your trouble occurs also there, if not I can share this vm with you if you want.Br,Sebastian

Hi Sebastian, I am also having trouble with the installation. Could you also share your VM to me? My email address is lzyang1995@outlook.com. Thanks a lot!

200729, 2019-10-07:
Hi,Sorry for late response. I will share today.

aliceqin12, 2019-10-16:
> Hi,Installation is so painfull that I would say just don't install it :PI prepared virtual machine Ubuntu 18.04 with pylayers installation I consider the most stable. I'll check if your trouble occurs also there, if not I can share this vm with you if you want.Br,Sebastian

Dear Sebastian, my installation for Pylayer went wrong as well:(
Could you kindly share the VM with me? Great thanks in advance!
My email address is aliceqin1212@gmail.com

DoHaiSon, 2024-02-19:
> Hi,Installation is so painfull that I would say just don't install it :PI prepared virtual machine Ubuntu 18.04 with pylayers installation I consider the most stable. I'll check if your trouble occurs also there, if not I can share this vm with you if you want.Br,Sebastian

Hi there and other guys,
I'm really looking for the VM image,
my email is: dohaison1998@gmail.com
Thank for your help!!!!

DoHaiSon, 2024-02-19:
> https://github.com/pylayers/pylayers/files/2542502/environment.txt is the file. I renamed it to .txt because .yml is not allowed as an upload.

There are many errors at this time on this yml file:
```
The following packages are incompatible
├─ geocoder ==1.38.1 py_0 does not exist (perhaps a typo or a missing channel);
├─ mkl ==2019.0 118 is requested and can be installed;
├─ mkl_fft ==1.0.6 py27h7dd41cf_0 is not installable because it requires
│  └─ mkl >=2018.0.3,<2019.0a0 , which conflicts with any installable versions previously reported;
└─ ratelim ==0.1.6 py27_0 does not exist (perhaps a typo or a missing channel).
```

FeilongZzz, 2024-03-27:
Dear Sebastian, I am so sorry for disturbing you after this long time. I would really appreciate it if you could also send me an image.. Thank you so much for your help! My email address is zhuang@miubiq.cs.titech.ac.jp. BEST REGARDS!

fejesmate, 2025-02-19:
> Hi,Installation is so painfull that I would say just don't install it :PI prepared virtual machine Ubuntu 18.04 with pylayers installation I consider the most stable. I'll check if your trouble occurs also there, if not I can share this vm with you if you want.Br,Sebastian

Hi Sebastian from 2025,
By any chance do you still have that vm image?
After spending a day trying to install things I'm at my wits end.
I would appreciate if you could share it with me at mat.fejes@gmail.com
Best Regards,
Mate