pylayers/pylayers #305: Coverage() error: local variable 'lonmin' referenced before assignment

johschmitz, 2018-03-16:
When trying to run the ipython notebook from

<http://pylayers.github.io/pylayers/auto_examples/plot_coverage.html#sphx-glr-auto-examples-plot-coverage-py>

on the current pylayers version it fails with the following error:
> ---------------------------------------------------------------------------
> UnboundLocalError                         Traceback (most recent call last)
> <ipython-input-2-17199c5718a6> in <module>()
>       3 import time
>       4 import pdb
> ----> 5 C = Coverage()
>       6 C.L._filename
>       7 C.tx = np.array((39,1))
> 
> /home/schmitz/work/python/pylayers/lib/python2.7/site-packages/pylayers-0.5-py2.7.egg/pylayers/antprop/coverage.pyc in __init__(self, _fileini)
>     138         else:
>     139             self.typ='outdoor'
> --> 140             self.E = ez.Ezone(filename)
>     141             self.E.loadh5()
>     142             self.E.rebase()
> 
> /home/schmitz/work/python/pylayers/lib/python2.7/site-packages/pylayers-0.5-py2.7.egg/pylayers/gis/ezone.pyc in __init__(self, prefix)
>     412         self.prefix = prefix
>     413 
> --> 414         (lom,loM,lam,laM) = dectile(self.prefix)
>     415 
>     416         self.extent = (lom,loM,lam,laM)
> 
> /home/schmitz/work/python/pylayers/lib/python2.7/site-packages/pylayers-0.5-py2.7.egg/pylayers/gis/ezone.pyc in dectile(prefix)
>     163         lonmax = lonmin+1
>     164 
> --> 165     return (lonmin,lonmax,latmin,latmax)
>     166 
>     167 def expand(A):
> 
> UnboundLocalError: local variable 'lonmin' referenced before assignment

niamiot, 2018-03-16:
It seems there is currently a issue in the eZone class, which manage outdoor scenes. But indoor is not involved in this bug. 

Which layout are you trying to use? (could you provide your **coverage.ini** file please ? ). Please check the .lay extension is present at the end of the file you try to use.

From my side, if I try to run the example you mention I 
do not face any issue with the current version of the code using the following coverage.ini file (in   pylayers_project/ini/coverage.ini  ):

```ini
[grid] 
nx = 80 
ny = 40 
boundary = [20,0,30,20] 
mode = full ; file, zone , full 
file = 'points.ini'

[layout] 
filename = TA-Office.lay ; 0 40 0 15 ;filename = W2PTIN.lay ;filename = Lstruc.lay

[ap] 
0 = {'name':'room1','wstd':'ieee80211b','p':(1,12,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Gauss','phideg':90} 
1 = {'name':'room2','wstd':'ieee80211b','p':(10,2,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Omni','phideg':0} 
2 = {'name':'room3','wstd':'ieee80211b','p':(20,1,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Gauss','phideg':75} 
3 = {'name':'room4','wstd':'ieee80211b','p':(36.5,1.5,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Gauss','phideg':120} 
4 = {'name':'room5','wstd':'ieee80211b','p':(25,12,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Gauss','phideg':180}

[rx] 
temperaturek = 300 
noisefactordb = 0

[show] 
show = True
```

johschmitz, 2018-03-16:
This is the code
```
from pylayers.antprop.coverage import *
import matplotlib.pyplot as plt
import time
import pdb
C = Coverage()
```
After that it crashes.
There is no coverage.ini file mentioned, so how can I find out which one I am using? I am basically trying to go with an unmodified default installation.
There must be some difference between your installation and my installation I think.

niamiot, 2018-03-16:
The default coverage.ini file used is located in the ini directory of your pylayers_project directory (which should be located in ~/pylayers_project. You can check at that path in the hidden file ~/.pylayers  ).

I reproduce the error only if I use a file without **.lay** extention in the filename parameter in the [layout] section of the ini file.

buguen, 2018-03-17:
Thank you  for your feedback, and sorry for the regression in the master branch.
Your problem is probably due to recent modification I did working on outdoor coverage. 

The new master version is probably fixing the problem.

johschmitz, 2018-03-21:
How can I understand which **coverage.ini** is being used by default when I am not specifying one?

niamiot, 2018-03-21:
by default Coverage uses the file "coverage.ini " from your pylayers_project directory. Otherwise, you can provide a configuration file as a parameter of the Coverage object.

johschmitz, 2018-03-21:
I am doing this to install pylayers:
`python setup.py install --prefix=<some_folder_in_home>/python/pylayers`

Then I end up with two files
`<home>/pylayers_project/ini/coverage.ini`
`<some_folder_in_home>/python/pylayers/data/ini/coverage.ini`

The second file is the one you are showing in your reply above. And the problem seems to be that the first file is different but taken by default. That leads to the crash. When I call the Coverage function with the second file as an argument then it works.

johschmitz, 2018-05-29:
> I'm getting the same problem (UnboundLocalError: local variable 'lonmin'
> referenced before assignment).
> Here's the full error:
> Traceback (most recent call last): File "example.py", line 6, in <module>
> C = Coverage() File
> "/home/USER/anaconda3/lib/python3.6/site-packages/pylayers-0.5-py3.6.egg/pylayers/antprop/coverage.py",
> line 143, in __init__ File
> "/home/USER/anaconda3/lib/python3.6/site-packages/pylayers-0.5-py3.6.egg/pylayers/gis/ezone.py",
> line 406, in __init__ File
> "/home/USER/anaconda3/lib/python3.6/site-packages/pylayers-0.5-py3.6.egg/pylayers/gis/ezone.py",
> line 157, in dectile UnboundLocalError: local variable 'lonmin' referenced
> before assignment
>
> I ran setup.py install with no path arguments, just like in installer_unix
>
> I've tried specifying the location for coverage.ini, but the same file
> then gave me the error
>
> Traceback (most recent call last): File "example.py", line 5, in <module>
> C = Coverage('~/pylayers_project/ini/coverage.ini') File
> "/home/USER/anaconda3/lib/python3.6/site-packages/pylayers-0.5-py3.6.egg/pylayers/antprop/coverage.py",
> line 98, in __init__ File
> "/home/USER/anaconda3/lib/python3.6/configparser.py", line 849, in items
> raise NoSectionError(section) configparser.NoSectionError: No section:
> 'layout'
> I'm probably overlooking something very simple, but I've tried
> johschmitz's fix to no avail and would appreciate if anyone else has an
> idea.

Did you go through all of these?

https://github.com/pylayers/pylayers/blob/master/INSTALL.md

Mornyres, 2018-05-30:
Unfortunately, yes. It did take a lot of adjustments to get things working with Python 3.6.4. If I had to guess, the problem likely lies somewhere in Python 2/3 compatibility. 

I was hoping to get it working on my main workstation for a quick demonstration, and then switch to 2.7 on a Pi or something if I were to actually use it on a project.

Mornyres, 2018-05-30:
Setting `lonmin`, `lonmax`, etc. to specific values before the conditional block is entered does get rid of the error, but presents a bunch of new errors depending on what I initialize them to.

For instance, 
```
lonmin = 0
lonmax = 180
latmin = 0
latmax = 90
```
 causes the error
`
GEOS_ERROR: b'TopologyException: Input geom 1 is invalid: Self-intersection at or near point 180.00000000018423 9.6296497219361793e-35 at 180.00000000018423 9.6296497219361793e-35'
`
While setting them to

```
lonmin = 1
lonmax = 90
latmin = 1
latmax = 90
```
results in no errors. I realize that this is just pushing the problem elsewhere, but pushing ahead nonetheless I then get the error

`
Traceback (most recent call last):
  File "example.py", line 5, in <module>
    C = Coverage('coverage.ini')
  File "/home/USER/anaconda3/lib/python3.6/site-packages/pylayers-0.5-py3.6.egg/pylayers/antprop/coverage.py", line 145, in __init__
  File "/home/USER/anaconda3/lib/python3.6/site-packages/pylayers-0.5-py3.6.egg/pylayers/gis/ezone.py", line 617, in rebase
AttributeError: 'Ezone' object has no attribute 'hgts'
`

```
        if source =='srtm':
->        Nlat,Nlon = self.hgts.shape                                  
        elif source=='aster':
          Nlat,Nlon = self.hgta.shape
```

buguen, 2018-05-31:
Dear all, 

Most of the problem comes from the transition from python 2.7 to python 3.5. 
Python 3.6 is challenging for having mayavi working properly. It is kind of tricky to find the good 
combination of packages. 

Below is my conda requirements : 
conda list --export > requirements.txt
# This file may be used to create an environment using:
# $ conda create --name <env> --file <this file>
# platform: linux-64
alabaster=0.7.10=py35_1
apptools=4.4.0=py35_0
asn1crypto=0.24.0=py35_0
babel=2.5.3=py35_0
backcall=0.1.0=py_0
basemap=1.1.0=py35_4
basemap-data-hires=1.1.0=0
blas=1.0=mkl
bleach=2.1.3=py_0
ca-certificates=2018.4.16=0
cairo=1.14.6=4
certifi=2018.4.16=py35_0
cffi=1.11.5=py35_0
chardet=3.0.4=py35_0
click=6.7=py_1
conda=4.5.1=py35_0
conda-env=2.6.0=0
configobj=5.0.6=py35_0
cryptography=2.2.1=py35_0
curl=7.59.0=1
cycler=0.10.0=py35_0
dask-core=0.17.2=py_0
decorator=4.3.0=py_0
descartes=1.0.2=py35_0
dill=0.2.7.1=py35_0
docutils=0.14=py35_0
entrypoints=0.2.3=py35_1
envisage=4.5.1=py35_0
expat=2.2.5=0
fastcache=1.0.2=py35_0
fontconfig=2.12.1=4
freetype=2.7=1
freexl=1.0.5=0
gdal=2.0.0=py35_1
geocoder=1.38.1=py_0
geos=3.6.2=1
geotiff=1.4.2=1
gettext=0.19.8.1=0
giflib=5.1.4=0
glib=2.51.4=0
gmp=6.1.2=0
gmpy2=2.0.8=py35_1
graphite2=1.3.11=0
h5py=2.7.1=py35_3
harfbuzz=1.4.3=0
hdf4=4.2.13=0
hdf5=1.10.1=2
html5lib=1.0.1=py_0
icu=58.2=0
idna=2.6=py35_1
imageio=2.3.0=py35_0
imagesize=1.0.0=py35_0
intel-openmp=2018.0.0=8
ipykernel=4.8.2=py35_0
ipython=6.3.1=py35_0
ipython_genutils=0.2.0=py35_0
ipywidgets=7.2.1=py35_1
jedi=0.12.0=py35_0
jinja2=2.10=py35_0
jpeg=9b=2
json-c=0.12.1=0
jsonschema=2.6.0=py35_1
jupyter=1.0.0=py_1
jupyter_client=5.2.3=py35_0
jupyter_console=5.2.0=py35_0
jupyter_contrib_core=0.3.3=py35_1
jupyter_contrib_nbextensions=0.5.0=py35_0
jupyter_core=4.4.0=py_0
jupyter_highlight_selected_word=0.2.0=py35_0
jupyter_latex_envs=1.4.4=py35_0
jupyter_nbextensions_configurator=0.4.0=py35_0
kealib=1.4.7=4
krb5=1.14.6=0
libdap4=3.18.3=2
libffi=3.2.1=3
libgcc=7.2.0=h69d50b8_2
libgcc-ng=7.2.0=hdf63c60_3
libgdal=2.0.0=1
libgfortran=3.0.0=1
libgfortran-ng=7.2.0=hdf63c60_3
libiconv=1.15=0
libnetcdf=4.4.1.1=10
libpng=1.6.34=0
libpq=9.6.3=0
libsodium=1.0.16=0
libspatialite=4.3.0a=18
libssh2=1.8.0=2
libstdcxx-ng=7.2.0=hdf63c60_3
libtiff=4.0.9=0
libxml2=2.9.8=0
libxslt=1.1.32=0
lime=0.1.1.18=py35_0
llvmlite=0.21.0=py35_0
lxml=4.2.1=py35_0
markupsafe=1.0=py35_0
matplotlib=2.0.0=np111py35_2
mayavi=4.5.0=py35_0
mistune=0.8.3=py35_1
mkl=2018.0.2=1
mkl_fft=1.0.2=py35_0
mkl_random=1.0.1=py35_0
mpc=1.1.0=4
mpfr=3.1.5=0
mpmath=1.0.0=py_0
multiprocess=0.70.5=py35_0
nbconvert=5.3.1=py_1
nbformat=4.4.0=py35_0
ncurses=5.9=10
netcdf4=1.3.1=py35_1
networkx=2.1=py35_0
notebook=5.4.1=py35_0
numba=0.36.1=py35hdc42b47_0
numpy=1.11.3=py35h3dfced4_4
numpydoc=0.8.0=py35_0
olefile=0.45.1=py35_0
openblas=0.2.20=7
openjpeg=2.1.2=3
openssl=1.0.2o=0
orderedset=2.0=py35_0
osmapi=1.1.0=py_0
packaging=17.1=py_0
pandas=0.22.0=py35_0
pandoc=2.1.3=0
pandocfilters=1.4.2=py35_0
pango=1.40.4=0
parso=0.2.0=py_0
pathos=0.2.1=py35_1
patsy=0.5.0=py35_0
pcre=8.39=0
pexpect=4.5.0=py35_0
pickleshare=0.7.4=py35_0
pillow=4.2.1=py35_0
pip=9.0.3=py35_0
pixman=0.34.0=1
pox=0.2.3=py35_0
ppft=1.6.4.7.1=py35_0
proj4=4.9.3=5
prompt_toolkit=1.0.15=py35_0
ptyprocess=0.5.2=py35_0
pycosat=0.6.3=py35_0
pycparser=2.18=py35_0
pyface=5.1.0=py35_0
pygments=2.2.0=py35_0
pyopenssl=17.5.0=py35_1
pyparsing=2.2.0=py35_0
pyproj=1.9.5.1=py35_0
pyqt=4.11.4=py35_3
pyserial=3.4=py35_0
pyshp=1.2.12=py_0
pyside=1.2.4=py35_8
pysocks=1.6.8=py35_1
pyspharm=1.0.9=py35_2
python=3.5.4=2
python-dateutil=2.7.2=py_0
pytz=2018.4=py_0
pywavelets=0.5.2=py35_1
pyyaml=3.12=py35_1
pyzmq=17.0.0=py35_4
qt=4.8.7=4
qtconsole=4.3.1=py35_0
ratelim=0.1.6=py35_0
readline=6.2=0
requests=2.18.4=py35_1
ruamel_yaml=0.15.35=py35_0
runipy=0.1.5=py35_1
scikit-image=0.13.1=py35_0
scikit-learn=0.19.1=py35hbf1f462_0
scipy=1.1.0=py35hfc37229_0
seaborn=0.8.1=py35h04cba02_0
send2trash=1.5.0=py_0
setuptools=39.0.1=py35_0
shapely=1.6.4=py35_0
simplegeneric=0.8.1=py35_0
simplejson=3.15.0=py35_0
simplekml=1.3.0=py35_0
sip=4.18=py35_1
six=1.11.0=py35_1
snowballstemmer=1.2.1=py35_0
sphinx=1.7.4=py35_0
sphinx-gallery=0.1.12=py35_0
sphinxcontrib-websupport=1.0.1=py35_0
sqlite=3.13.0=1
statsmodels=0.8.0=py35_0
sympy=1.1.1=py35_0
terminado=0.8.1=py35_0
testpath=0.3.1=py35_0
tk=8.5.19=2
toolz=0.9.0=py_0
tornado=5.0.2=py35_0
tqdm=4.23.1=py_0
traitlets=4.3.2=py35_0
traits=4.5.0=py35_0
traitsui=5.1.0=py35_0
typing=3.6.4=py35_0
urllib3=1.22=py35_0
util-linux=2.21=0
vtk=7.0.0=py35_0
wcwidth=0.1.7=py35_0
webencodings=0.5=py35_0
wheel=0.31.0=py35_0
widgetsnbextension=3.2.1=py35_0
xerces-c=3.2.0=0
xz=5.2.3=0
yaml=0.1.7=0
zeromq=4.2.5=1

Paste those requirements in requirements.txt 

then 

conda create --name <envname> --file requirements.txt

buguen, 2018-05-31:
This is a working version executed on 31 May with the code on the master branch. 
[Plot Coverage example](https://pylayers.github.io/pylayers/auto_examples/plot_coverage.html#sphx-glr-auto-examples-plot-coverage-py)

closed 2018-05-31 by buguen