pylayers/pylayers #317: Coverage Problem - Error reading grid mode

sahibdhanjal, 2019-01-24:
Hi, I was trying to get up and running with the pylayers coverage example notebook [here](https://pylayers.github.io/pylayers/auto_examples/plot_coverage.html#sphx-glr-auto-examples-plot-coverage-py), however, when trying to run it directly, it gave me the following errors

[image]

As per #305 I, checked my coverage.ini file, and replaced it with what was mentioned as follows. 

```
[grid]
nx = 80
ny = 40
boundary = [20,0,30,20]
mode = full ; file, zone , full
file = 'points.ini'

[layout]
filename = TA-Office.lay ; 0 40 0 15 ;filename = W2PTIN.lay ;filename = Lstruc.lay

[ap]
0 = {'name':'room1','wstd':'ieee80211b','p':(1,12,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Gauss','phideg':90}
1 = {'name':'room2','wstd':'ieee80211b','p':(10,2,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Omni','phideg':0}
2 = {'name':'room3','wstd':'ieee80211b','p':(20,1,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Gauss','phideg':75}
3 = {'name':'room4','wstd':'ieee80211b','p':(36.5,1.5,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Gauss','phideg':120}
4 = {'name':'room5','wstd':'ieee80211b','p':(25,12,1.2),'PtdBm':0,'chan':[11],'on':True,'ant':'Gauss','phideg':180}

[rx]
temperaturek = 300
noisefactordb = 0

[show]
show = True
```
I'm trying to run an out of the box installation after installing all the requirements as per the installation doc, however, can't seem to figure out the issue. How can I get this working?

buguen, 2019-01-26:
Hi Sahib, 

The reason is that you cannot read the File TA-Office.lay because of the continuation comment which is not properly handled by the reader. A simple fx is to remove the comment :
 ; 0 40 0 15 ;filename = W2PTIN.lay ;filename = Lstruc.lay

TODO fix the comment on the same line issue

niamiot, 2019-01-26:
@buguen

It appears this is a configparser option to activate
https://docs.python.org/3/library/configparser.html

inline_comment_prefixes, default value: None

Comment prefixes are strings that indicate the start of a valid comment within a config file. comment_prefixes are used only on otherwise empty lines (optionally indented) whereas inline_comment_prefixes can be used after every valid value (e.g. section names, options and empty lines as well). By default inline comments are disabled and '#' and ';' are used as prefixes for whole line comments.

Changed in version 3.2: In previous versions of configparser behaviour matched comment_prefixes=('#',';') and inline_comment_prefixes=(';',).

sahibdhanjal, 2019-01-27:
Removing the comments fixed it, thanks! Closing this issue

closed 2019-01-27 by sahibdhanjal