pylayers/pylayers #320: Layout Dimensions

Docopoper, 2019-02-07:
Hey, are the coordinates of the points in a layout file measured in meters?

Docopoper, 2019-02-07:
Oh, also, where did you get your values for material properties?
If there's a database of those values that'd be super useful.

Thanks. (I am using PyLayers for an experiment for a paper, so I shall reference yours)

buguen, 2019-02-09:
The layout format contains an info section at the beginning of the 
[info]
format = cart 
version = 1.3
typ = indoor

If the format is set to cart (cartesian) then all dimensions are in meters. Another format could be latlon with points defined with their (lat,lon,height) coordinates.

buguen, 2019-02-09:
For the material properties, a good start is to use the values from the ITU recommendation 

https://www.itu.int/dms_pubrec/itu-r/rec/p/R-REC-P.2040-1-201507-I!!PDF-E.pdf

All the ITU materials defined in table 3 page 23 from the ITU document above are predefined in the __init__ of Mat class. To use 
it you have to prefix the material nae with ITU. (see below)

See also directly in the code in Slab.py module 
https://github.com/pylayers/pylayers/blob/master/pylayers/antprop/slab.py

Any idea for a better implementation of useful features for Slab and Material is more than welcome. 

 #MATERIAL ;  a  ; b ;   c  ;  d  ;  fmin ; fmax 

      
        # ITU-R P2040 Table 3
        ITU_P2040_T3 = {
        'ITU_CONCRETE' : np.array([  5.31  ,  0 , 0.0326 , 0.8095, 1, 100]),
        'ITU_BRICK' : np.array([ 3.75 , 0 , 0.038 , 0 , 1 , 10]),
        'ITU_PLASTERBOARD' : np.array([ 2.94 , 0 , 0.0116 , 0.7076 , 1 , 100]),
        'ITU_WOOD' : np.array([ 1.99 , 0 , 0.0047 , 1.0718 , 0.001 , 100]),
        'ITU_GLASS' : np.array([ 6.27 , 0 , 0.0043 , 1.1925 , 0.1 , 100]),
        'ITU_CEILINGBOARD' : np.array([ 1.50 , 0 , 0.0005 , 1.1634 , 1 , 100]),
        'ITU_CHIPBOARD' : np.array([ 2.58 , 0 , 0.0217 , 0.7800 , 1 , 100]),
        'ITU_FLOORBOARD' : np.array([ 3.66 , 0 , 0.0044 , 1.3515 , 50 , 100]),
        'ITU_VERYDRYGROUND' : np.array([ 3 , 0 , 0.00015 , 2.52 , 1 , 10]),
        'ITU_MEDIUMDRYGROUND' : np.array([ 15 , -0.1 , 0.035 , 1.63 , 1 , 10]),
        'ITU_WETGROUND' : np.array([ 30 , -0.4 , 0.15 , 1.30 , 1 , 10])
        }
        # Parameters a,b,c,d 
        # Table 3 Rec ITU-R P.2040.1
        #
        # epsrp  = a * fGHZ ** b 
        # sigma  = c * fGHZ ** d 
        #
        if (name in ITU_P2040_T3.keys()):
            abcd = ITU_P2040_T3[name]
            self['a'] = abcd[0]
            self['b'] = abcd[1]
            self['c'] = abcd[2]
            self['d'] = abcd[3]

The description of slabs and materials has to be included in the .lay file. 
below is an example 

[slabs]
 CONCRETE_TC1_METIS = {'color': 'black', 'linewidth': 2, 'lthick': [0.108], 'lmatname': ['CONCRETE_TC1_ME    TIS']}
 _AIR = {'color': 'white', 'linewidth': 1, 'lthick': [0.02], 'lmatname': ['AIR']}
 WOOD_TC1_METIS = {'color': 'brown', 'linewidth': 1, 'lthick': [0.019], 'lmatname': ['WOOD_TC1_METIS']}
 AIR = {'color': 'white', 'linewidth': 1, 'lthick': [0.02], 'lmatname': ['AIR']}
 CEIL = {'color': 'grey20', 'linewidth': 1, 'lthick': [0.1], 'lmatname': ['REINFORCED_CONCRETE']}
 FLOOR = {'color': 'grey40', 'linewidth': 1, 'lthick': [0.1], 'lmatname': ['REINFORCED_CONCRETE']}

  [materials]
  CONCRETE_TC1_METIS = {'mur': 1, 'epr': (6.14-0.3j), 'roughness': 0.0, 'sigma': 0.0}
  AIR = {'mur': (1+0j), 'epr': (1+0j), 'roughness': 0.0, 'sigma': 0.0}
  WOOD_TC1_METIS = {'mur': 1, 'epr': (1.64-0.11j), 'roughness': 0.0, 'sigma': 0.0}
  REINFORCED_CONCRETE = {'sigma': 3.0, 'roughness': 0.0, 'epr': (8.69999980927+0j), 'mur': (1+0j)}

There are several manners to define material in PyLayers, one of them is to use the ITU parameters.
The best way is to look in the code at that 

I agree that it would be super useful to gather in a database all the different value which have been used across the literature. This is definitely an interesting feature that could be proposed in a near future.