pylayers/pylayers #326: networkx error

AmanAmir, 2019-05-21:
Hello
I am new to python and pylayers and trying to run the example simulation for ray tracing transmission channel simulation and get this error
File "/home/FE/memon/anaconda3/lib/python3.7/site-packages/networkx/convert_matrix.py", line 765, in to_scipy_sparse_matrix
raise nx.NetworkXError("Graph has no nodes or edges")

NetworkXError: Graph has no nodes or edges
The problem is with the L.Build() command.
Any help on how to run this ?

buguen, 2019-05-29:
Could you be more specific on the condition of your simulation. 
Which layout file do you use ?

AmanAmir, 2019-05-29:
I am trying to run the example code given on this link https://pylayers.github.io/pylayers/notebook/2-AP/Channel.html.I get the error run on the line L.build()

It also fails to import the module simulem. I comment that out is it because of that? I do not know.

AmanAmir, 2019-05-29:
runfile('/home/FE/memon/channel.py', wdir='/home/FE/memon')
Traceback (most recent call last):

  File "<ipython-input-15-0c40d85a3fbe>", line 1, in <module>
    runfile('/home/FE/memon/channel.py', wdir='/home/FE/memon')

  File "/home/FE/memon/anaconda3/lib/python3.7/site-packages/spyder_kernels/customize/spydercustomize.py", line 786, in runfile
    execfile(filename, namespace)

  File "/home/FE/memon/anaconda3/lib/python3.7/site-packages/spyder_kernels/customize/spydercustomize.py", line 110, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/home/FE/memon/channel.py", line 17, in <module>
    from pylayers.simul.simulem import *

ModuleNotFoundError: No module named 'pylayers.simul.simulem'