pylayers/pylayers #328: Error in algebraic.py: global name 'rg_ramerstd' is not defined

lquinet, 2019-05-29:
Hi, 

When I tried to execute the file test_algebraic.py in pylayers/pylayers/location/algebraic/tests , I have an error:

```
Traceback (most recent call last):
  File "test_algebraic.py", line 52, in <module>
    print S.crb(nodes['BN'],0,1,0)
  File "build\bdist.win-amd64\egg\pylayers\location\algebraic\algebraic.py", line 1145, in crb
  File "build\bdist.win-amd64\egg\pylayers\location\algebraic\algebraic.py", line 1084, in fim
NameError: global name 'rg_ramerstd' is not defined   
```

I suspect there is an error in fim method of algloc class in algebraic.py, because we access the variable rg_ramerstd[i] which is no initialised anywhere. This error happens only when tdoa=1 in the parameter of the method.

Or perhaps something is misconfigured in my python environment? I'm on Windows 10 and I'm using an anaconda environment in python 2.7. Here is the list of the packages installed:

```
# packages in environment at C:\Users\Utilisateur\Anaconda3\envs\pylayers:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py27_0
apptools                  4.4.0                    py27_1
asn1crypto                0.24.0                   py27_0
astroid                   1.6.5                    py27_0
attrs                     19.1.0                   py27_1
babel                     2.6.0                    py27_0
backports                 1.0                      py27_1
backports.functools_lru_cache 1.5                      py27_1
backports.shutil_get_terminal_size 1.0.0                    py27_2
backports_abc             0.5                        py_0
basemap                   1.2.0                    pypi_0    pypi
bitstring                 3.1.5                    pypi_0    pypi
blas                      1.0                         mkl
bleach                    3.1.0                    py27_0
bzip2                     1.0.6                h0c8e037_5
ca-certificates           2019.3.9             hecc5488_0    conda-forge
certifi                   2019.3.9                 py27_0    conda-forge
cffi                      1.12.3           py27hcfb25f9_0
cftime                    1.0.3.4          py27hc997a72_0
chardet                   3.0.4                    pypi_0    pypi
click                     7.0                      pypi_0    pypi
cloudpickle               1.0.0                      py_0
colorama                  0.4.1                    pypi_0    pypi
configobj                 5.0.6                    py27_1
configparser              3.7.4                    py27_0
cryptography              2.4.2            py27hc64555f_0
curl                      7.62.0               h7a46e7a_0
cycler                    0.10.0           py27h59acbbf_0
cython                    0.29.7                   pypi_0    pypi
decorator                 4.4.0                    py27_1
defusedxml                0.6.0                      py_0
descartes                 1.1.0                    pypi_0    pypi
dill                      0.2.9                    pypi_0    pypi
django                    1.11.20                  pypi_0    pypi
docutils                  0.14             py27h8652d09_0
entrypoints               0.3                      py27_0
enum34                    1.1.6                    pypi_0    pypi
envisage                  4.7.1                      py_0
expat                     2.2.5                h2880e7c_0
freetype                  2.9.1                h4d385ea_1
freexl                    1.0.5                h0c8e037_0
functools32               3.2.3.2                  py27_1
future                    0.17.1                   pypi_0    pypi
futures                   3.2.0                    py27_0
gdal                      2.0.0                    py27_1
geocoder                  1.38.1                   pypi_0    pypi
geos                      3.6.2                heec837b_2
h5py                      2.8.0            py27h5f1a774_2
hdf4                      4.2.13               hd0a9391_2
[... 127 lines cut ...]
traitlets                 4.3.2                    pypi_0    pypi
traits                    5.1.1            py27h0c8e037_0
traitsui                  6.0.0                    py27_1
triangle                  20170429                 pypi_0    pypi
typing                    3.6.6                    py27_0
unittest2                 1.1.0                    py27_0
urllib3                   1.25.3                   pypi_0    pypi
vc                        9                    h7299396_1
vs2008_runtime            9.00.30729.1         hfaea7d5_1
vtk                       6.3.0                    py27_1
wcwidth                   0.1.7                    pypi_0    pypi
webencodings              0.5.1                    py27_1
wheel                     0.33.4                   py27_0
win-unicode-console       0.5                      pypi_0    pypi
win_inet_pton             1.1.0                    py27_0
win_unicode_console       0.5              py27hc037021_0
wincertstore              0.2              py27hf04cefb_0
wrapt                     1.11.1           py27h0c8e037_0
wxpython                  4.0.4            py27hc56fc5f_0
xerces-c                  3.2.2                hc56fc5f_0
xz                        5.2.4                h3cc03e0_4
zeromq                    4.3.1                h2880e7c_3
zlib                      1.2.11               h3cc03e0_3
zstd                      1.3.7                h1b0e4d7_0
```

buguen, 2019-05-29:
I have fixed the error. It should work now.

lquinet, 2019-05-31:
Tested in python 3.6, it works.

closed 2019-05-31 by lquinet