pylayers/pylayers #330: TDOA localisation in 3D

lquinet, 2019-05-31:
Hi,

I would like to use the TDOA localisation in 3D. The only information I measure is the time difference between blind node (tag) and reference nodes (anchors). I have tried algebraic and geometric localisation. 

In algebraic localisation it only works well when I use 5 reference nodes. Are you using a linear method that involve the use of minimum 5 reference nodes?

In geometric localisation it only works well in 2D but not in 3D, even if I use 5 reference nodes.

Is it possible to have access to the theory used for the localisation, especially for algebraic localisation? 

Thanks in advance.

smbat-jalalyan, 2019-09-07:
I also opened issue #336 related to this. I was wondering if you maybe resolved your issue or came up with a workaround?

lquinet, 2019-09-09:
Hi,

Although Pylayers is very powerful and versatile, I gave up it because it is not enough supported and documented. I didn't resolved my issue but I'm using this library to do TDOA: https://github.com/AlexisTM/MultilaterationTDOA .