pylayers/pylayers #351: Creating a new layout file

reenaraj073, 2020-06-17:
Hello everyone, I have created a new layout file for indoor environment by writing into a new .lay file , just by understanding from the already present .lay files.
My doubts:

1. Is this the right way/efficient way to define the layout file?
2. Does newly created .lay files have access to slab.ini and mat.ini files just like currently existing files? If yes?, how can i make this file access the slab and .ini files? 

Any kind of help is appreciated...

horvlint, 2020-10-16:
Hello,
I found several alternatives for creating layout files:
- pylayers/gui/editor.py tool (though I found some portability issues if using with python3
- java open street map: https://josm.openstreetmap.de the layouts created with this can be imported to pylayers and saved as a .lay file

reenaraj073, 2020-10-21:
> Hello,
> I found several alternatives for creating layout files:
> 
> * pylayers/gui/editor.py tool (though I found some portability issues if using with python3
> * java open street map: https://josm.openstreetmap.de the layouts created with this can be imported to pylayers and saved as a .lay file

Hi, thank you for your recommendations. I was not aware of the 1st option you provided. I will check this out.

closed 2020-10-21 by reenaraj073