pylayers/pylayers #354: install lib

shihchun, 2020-11-12:
I've install the shapely, Basemap... with but still have some prob with python 3.8 in win10,

```
install note....
 Pip install Mayavi shapely Basemap network

Basemap installation
pip install egos
cp3.8 python version
https://www.lfd.uci.edu/~gohlke/pythonlibs/

import os
os.environ['PROJ_LIB'] = r'/Users/huangshijun/anaconda3/share/proj'

Shapely installation
cp3.8 python version
https://towardsdatascience.com/install-shapely-on-windows-72b6581bb46c

descartes
pip install descarteslabs triangle meshpy  osmapi simplejson pdbpp
pip install smopy
conda install -c conda-forge simplekml 
pip install networkx==2.0
pip install matplotlib==3.1.2
pip install scipy==1.2.1

Mayavi
conda install -c conda-forge vtk 
Pip Install  vtk_cp38.whl
conda install -c conda-forge mayavi 

********************************************************************************
WARNING: Imported VTK version (9.0) does not match the one used
         to build the TVTK classes (8.2). This may cause problems.
         Please rebuild TVTK.
********************************************************************************
reinstall VTK to 8.2 download vtk8.2**.whl
pip install vtk8.2**.whl

import sys
sys.setdefaultencoding('UTF8')

改成

import imp imp.reload(sys)
export PYTHONIOENCODING=utf8
```

I need help, could some provide the requirement.txt ?

```
python --version > pyversion.txt
pip freeze > requirements.txt
```

shihchun, 2020-11-12:
exLuebber.py
debug in python 3.8

[image]

add the `list()`
[image]

[image]

shihchun, 2020-11-16:
plot_coverage.py

[image]
[image]

KeyError: zfloor & zceil change to floor & ceil
[image]

```python
python setup.py clean
python setup.py build
python setup.py install
```

[image]

shihchun, 2020-11-16:
```
python --version > pyversion.txt
pip freeze > requirements.txt
conda list --explicit > spec-file.txt
```
[requirements.txt](https://github.com/pylayers/pylayers/files/5544392/requirements.txt)
[spec-file.txt](https://github.com/pylayers/pylayers/files/5544393/spec-file.txt)

cdd017, 2020-11-16:
hi ... Shishcun 
 i try to create conda env  as you describe in the  spec-file.
got this message :
Preparing transaction: done
Verifying transaction: failed

PaddingError: Placeholder of length '29' too short in package /home/dannyd/anaconda3/envs/pyl385/Library/bin/tcl86t.dll.
The package must be rebuilt with conda-build > 2.0. 

Can you Help Pleas ??
[image]

shihchun, 2020-11-16:
I don't know about it, just export it = = 
Sorry for that

shihchun, 2020-11-16:
[image]
You need to download them and install with

```sh
pip install /path_to_file/file_name
```

shihchun, 2020-11-16:
@cdd017 
[image]
I guess you can try this with yml file

shihchun, 2020-11-16:
`\pylayers\pylayers\antprop\spharm.py`

all `scipy.misc.factorial` change to `scipy.special.factorial`

```
# from scipy.misc
# to
from scipy.special import factorial
```

cdd017, 2020-11-16:
Hi 
pleas can you create env YAML file  so i can clone the environment

shihchun, 2020-11-16:
I already upload it. as environment.txt, check it out.

shihchun, 2020-11-18:
@cdd017  

You can test this env on other issue I found, they test at 2018

[issue296](https://github.com/pylayers/pylayers/issues/296)

[environmentBase_cp36.yaml](https://github.com/shihchun/pylayers/blob/master/environmentBase_cp36.yaml)

shihchun, 2020-11-18:
@cdd017 

I check the install on wsl Linux and my MacBook, I found the vtk install is complicated. I recommended you install with win10 + anaconda with the whl package. cuz the vtk need to build from source, just like the gccxml in ns3.  (= =)

I try the `environment.yaml`, the reason you can't create environment is cuz of the dependency version not support on your machine. you need add `#` to mark that line, after doing so,  it might not the env you want.

```sh
conda env create -f environmentBase_cp36.yaml
Collecting package metadata (repodata.json): done
Solving environment: failed

ResolvePackageNotFound:
  - ruamel_yaml==0.15.46=py36h14c3975_0
  - jbig==2.1=hdba287a_0
  - scipy==1.1.0=py36hfa4b5c9_1
  - mpc==1.1.0=h10f8cd9_1
  - kiwisolver==1.0.1=py36hf484d3e_0
```

I provide my note to you. I not actually build from source.

# install note
open powershell with admin install chocolatey copy from their site

```sh
Set-ExecutionPolicy Bypass -Scope Process -Force; [System.Net.ServicePointManager]::SecurityProtocol = [System.Net.ServicePointManager]::SecurityProtocol -bor 3072; iex ((New-Object System.Net.WebClient).DownloadString('https://chocolatey.org/install.ps1'))
```

install the display for wsl

```sh
choco install vcxsrv # install on windows open app bind the port 0 on admin cmd.exe
sudo apt-get install x11-apps # cmd type wsl to ubuntu
echo "export DISPLAY=127.0.0.1:0" >> ~/.bashrc  # wsl display bind port for x11 server
```

```sh
git clone pylayers
cd pylayers
./installer_unix
python setup.py clean
python setup.py build
python setup.py install
```

```sh
python ant.py
Layout:Mayavi is not installed # get error
```

I check the install package, the [vtk](https://pypi.org/project/vtk/#history) package on python pip, all the support after `vtk8.x`, and the libvtk version need to as same as the pip vtk version. (whl package is easy a lot !)

the apt-get test untial `libvtk6`

```sh
#sudo apt-get install libvtk6-dev #  you need to build from source 
# install the tool to build
sudo apt install cmake libavcodec-dev libavformat-dev libavutil-dev libboost-dev libdouble-conversion-dev libeigen3-dev libexpat1-dev libfontconfig-dev libfreetype6-dev libgdal-dev libglew-dev libhdf5-dev libjpeg-dev libjsoncpp-dev liblz4-dev liblzma-dev libnetcdf-dev libnetcdf-cxx-legacy-dev libogg-dev libpng-dev libpython3-dev libqt5opengl5-dev libqt5x11extras5-dev libsqlite3-dev libswscale-dev libtheora-dev libtiff-dev libxml2-dev libxt-dev qtbase5-dev qttools5-dev zlib1g-dev
git clone git@gitlab.kitware.com:vtk/vtk.git
cd vtk
git checkout v8.2.0
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=$HOME/vtk-inst \
    -DCMAKE_INSTALL_RPATH=$HOME/vtk-inst \
    -DVTK_Group_Qt=ON \
    -DVTK_QT_VERSION=5 \
    -DVTK_Group_Imaging=ON \
    -DVTK_Group_Views=ON \
    -DModule_vtkRenderingFreeTypeFontConfig=ON \
    -DVTK_WRAP_PYTHON=ON \
    -DVTK_PYTHON_VERSION=3 \
    -DPYTHON_EXECUTABLE=/usr/bin/python3 \
    -DPYTHON_INCLUDE_DIR=/usr/include/python3.6 \
    -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so \
    -DBUILD_TESTING=OFF \
    -DVTK_USE_SYSTEM_LIBRARIES=ON \
    -DVTK_USE_SYSTEM_LIBPROJ4=OFF \
    -DVTK_USE_SYSTEM_GL2PS=OFF \
    -DVTK_USE_SYSTEM_LIBHARU=OFF \
    -DVTK_USE_SYSTEM_PUGIXML=OFF \
    -DCMAKE_BUILD_TYPE=Release \
    ..
make -j$(($(nproc) - 1))
make install
```

```sh
pip install vtk mayavi
sudo apt-get install python3-pyqt5
pip install vtk PyQt5 mayavi shapely

sudo apt-get install mayavi2

conda install mayavi
pip install mayavi shapely
```

closed 2020-11-21 by shihchun

cdd017, 2020-11-21:
Many thank for your efforts..
Ill try tomorrow morning to set it up ..
Thanks