pylayers/pylayers #357: problem running code

cdd017, 2020-11-19:
i try to run example code from the site 

i receive errors  for any import command

n [11]: runcell(0, 'C:/Users/dannyd/untitled0.py')
Traceback (most recent call last):

  File "C:\Users\dannyd\untitled0.py", line 8, in <module>
    from pylayers.gis.ezone import *

  File "<frozen zipimport>", line 259, in load_module

  File "C:\Users\dannyd\Anaconda3\envs\pyl38\lib\site-packages\pylayers-0.5-py3.8.egg\pylayers\gis\ezone.py", line 24, in <module>
    import pylayers.util.geomutil as geu

  File "<frozen zipimport>", line 259, in load_module

  File "C:\Users\dannyd\Anaconda3\envs\pyl38\lib\site-packages\pylayers-0.5-py3.8.egg\pylayers\util\geomutil.py", line 25, in <module>
    from descartes.patch import PolygonPatch

ModuleNotFoundError: No module named 'descartes'

Pleas Help.

shihchun, 2020-11-19:
pip install descartes

cdd017, 2020-11-19:
No success....after pip install descartes

>>> from pylayers.gis.ezone import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\dannyd\Anaconda3\envs\pyl38\lib\site-packages\pylayers-0.5-py3.8.egg\pylayers\gis\ezone.py", line 26, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\dannyd\Anaconda3\envs\pyl38\lib\site-packages\pylayers-0.5-py3.8.egg\pylayers\antprop\loss.py", line 19, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\dannyd\Anaconda3\envs\pyl38\lib\site-packages\pylayers-0.5-py3.8.egg\pylayers\gis\gisutil.py", line 4, in <module>
ModuleNotFoundError: No module named 'simplejson'

cdd017, 2020-11-19:
Can you address me how to run a sample program? 
pleas

shihchun, 2020-11-19:
pip install simplejson

shihchun, 2020-11-19:
Maybe you can try this 

```python
from pylayers.simul.link import *
DL = DLink(L='defstr.lay')
DL.eval()
DL._show3()
```

cdd017, 2020-11-19:
(pyl38) C:\Users\dannyd\pylayers-master>python
Python 3.8.3 (default, Jul  2 2020, 17:30:36) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
>>>
>>>
>>>
>>>
>>> from pylayers.simul.link import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\dannyd\pylayers-master\pylayers\simul\link.py", line 30, in <module>
    from pylayers.simul.radionode import RadioNode
  File "C:\Users\dannyd\pylayers-master\pylayers\simul\radionode.py", line 26, in <module>
    from pylayers.mobility.trajectory import *
  File "C:\Users\dannyd\pylayers-master\pylayers\mobility\trajectory.py", line 30, in <module>
    from pylayers.gis.layout import Layout
  File "C:\Users\dannyd\pylayers-master\pylayers\gis\layout.py", line 32, in <module>
    import triangle
ModuleNotFoundError: No module named 'triangle'

shihchun, 2020-11-19:
pip install triangle (==

cdd017, 2020-11-19:
I'm running on windows,
pip install triangle already did. but no success

shihchun, 2020-11-19:
download the whl [package ](https://www.lfd.uci.edu/~gohlke/pythonlibs/)
[image]