pylayers/pylayers #359: need help for using evalH

mostofi386, 2020-11-23:
hi. I need to extract the channel transfer function. Can anyone please giving me an example: how to use the evalH function and extract the channel transfer function as a matrix.
I will be appreciated for your reply, thanks!

shihchun, 2020-11-24:
```python
from pylayers.simul.link import *
import matplotlib.pyplot as plt

L=Layout('defstr.lay')
rx = np.array([8,4,1.6])
tx = np.array([2,3,1.5])
fGHz = np.linspace(2,6,500)
Aa = Antenna('Omni',fGHz=fGHz)
Ab = Antenna('Omni',fGHz=fGHz)

L.build()
L.dumpw() # <= This is the trick
L._showGi()
Lk = DLink(L=L,a=tx,b=rx,Aa=Aa,Ab=Ab,fGHz=fGHz)
Lk.eval(force=True)
Lk._show3() # Not sure 3d works on wsl and macOS or not
# other thing may need check the code
Lk.H # return Tchannel pylayers\pylayers\antprop\channel.py
Lk.H.y
```

`Lk.H.y` is looks like the channel state information

shape is rays_ammout x Nr x Nt x Nf

I wanna get the channel matrix too.

mostofi386, 2020-11-24:
thanks for your reply. Sorry, I'm not really good in python. what do you mean with sth?

shihchun, 2020-11-24:
[image] - from WiFi Sensing with Channel State Information: A Survey

I guess it's CSI estimate, like this fig I found in the paper, we may need to do the CSI measurement.
I think we need to try with source code, or wait for author's response = =, There still have a lot little bug
You might need to  do some research, and then try the matrix with your system.

I just try with the Lk.H.y, Nt=1, Nr=1, but get a little bar, so I guess it's CSI.

mostofi386, 2020-12-03:
actually, I'm still confused and looking for the right answer that might give me the real channel matrix :(
I would appreciate if anyone can help me with this function

shihchun, 2020-12-03:
I guess you can use closed loop or open loop channel estimate method. Test the channel capacity.  At least 1000 Lk.eval() in different position. It’s not easy for me.

I try the NYU sim, but the capacity is strange, and the MATLAB code only generate the when Nt=8, Nr=4, and u need to press the button and only get one moment matrix. I think you can play this one. If you have result plis let me know. It always not works for me.

For example here, suppose it’s NLOS  Channel likes the Rayleigh BW 20MHz，speed less than 75Mbps, LOS channel like the Rican...etc check the paper value.

If the value looks fine it might works 

https://github.com/XassassinXsaberX/communication-simulation/blob/master/MIMO/channel%20capacity/open-loop%20%26%20close-loop%20channel%20capacity.py