pylayers/pylayers #360: problem with ray tracing example

mostofi386, 2020-12-04:
Hi everyone,

I tried simple ray tracing but there is an issue that I cant fix it. Does anyone know how it will be shown correctly?
>>> from pylayers.simul.link import *
>>> DL = DLink(L='defstr.lay')
>>> DL.eval()
>>> DL._show3()
[image]

also, can anyone help me understand how can I change Nr and Nt in this example?
Thank you

shihchun, 2020-12-05:
Try this `/pylayers/simul/tests/test_MIMO3.py`
But I can’t see 3D plot with this file.

mostofi386, 2020-12-07:
I don't get it. How is it related to my issue?

shihchun, 2020-12-07:
https://github.com/pylayers/pylayers/blob/master/pylayers/simul/tests/test_MIMO3.py

This is the MIMO example in the test file I found.

The file call other function to build antenna array obj.

shihchun, 2020-12-08:
[image]
It's Antenna Array, I don't know how it works. You can try it

mostofi386, 2020-12-08:
I tried test_MIMO3 but still, I get the same shape. Do you what is the problem?

shihchun, 2020-12-08:
That’s your prob I think, it return the antenna array.

` ``python
cd to/filePATH
ipython
%run test_MIMO3.py
Lk.H
H # baseband
` ``

mostofi386, 2020-12-08:
It was already like that. I mean in the example test_MIMO3.
but with this function(DL._show3() ), I get this error:

ValueError: operands could not be broadcast together with shapes (20,100) (1,20)

and there is no plot in 3D

shihchun, 2020-12-09:
I already told you I have this prob

shihchun, 2020-12-09:
This shape I guess may good for research of the CSI, I not familiar with it. For the equalization between channel and transmitter, the closed loop system use CSIT as feedback. In MU-MIMO estimate from AOA angle with TOF time.

You have x=symbol u sent. i guess it’s impulse function or rect you sent as “wav” property, the source shows the default value.

If it’s input, and you have y = Hs+n (suppose it is), and u have doa dod, u can estimate a lot things. You have that angle you can estimate the CSI, just like the SDR, but you have ray signatures....etc.

But the SINR and power gain ....etc. you need to think when u do the simulations, it’s painful for me.

mostofi386, 2020-12-09:
I will try this. thank you so much for your time and answers.

themag93, 2021-03-30:
Hi @mostofi386,
I'm having the exact same issues as you over here, have you been able to solve them by now? Actually, I don't understand a word of what @shihchun is trying to say.

shihchun, 2021-03-31:
That’s complicated for me to check the source code, and try. I not sure what inside the class under this project. Even the test script there, I don’t know it reliable or not.

mostofi386, 2021-03-31:
> Hi @mostofi386,
> I'm having the exact same issues as you over here, have you been able to solve them by now? Actually, I don't understand a word of what @shihchun is trying to say.

hi. I couldn't solve it and I just give on Pylayers.

themag93, 2021-03-31:
Thanks for the update, to both of you! If you know any other tool for indoor channel modeling with similar features (i e. room layout modeling, moving actors ..) I'd be glad for a recommendation!

shihchun, 2021-03-31:
This is a tools using  3gpp38901 propagation model, propagation model roughly three type pathloss, large scale, small scale, it happened in the same time, but we usually simulate with one of it. 

https://github.com/Sonohi/monster/tree/master/examples/PowerSINRmap/figures

https://github.com/Sonohi/monster/tree/master/channel

if u wanna simulate physical layer, hata model is ok enough, u don’t need any tool

https://www.gaussianwaves.com/2019/03/hata-okumura-model-for-outdoor-propagation