pylayers/pylayers #361: Problem of running the code in "Example of issue 314"

srisooksai, 2020-12-24:
I followed the instruction in ["Example of issue 314"](http://pylayers.github.io/pylayers/auto_examples/plot_issue314.html) by running this code.

`from pylayers.gis.layout import *`
`L = Layout((57.696495, 12.006234),dist_m=100)`
`L.showG('s')`
`plt.show(block=True)`

Then I got the "lat_12_0095745_lon_57_695605.lay" file (details are shown at the bottom).
However, when I run this code,

`L = Layout('lat_12_0095745_lon_57_695605.lay')`

I got this error.
> AssertionError: None
> /home/kyosan/anaconda3/envs/conda_pylayers/lib/python3.6/site-packages/pylayers-0.5-py3.6.egg/pylayers/gis/layout.py(1275)g2npy()
>    1273 
>    1274             scale = np.sqrt(normx * normx + normy * normy)
> -> 1275             assert (scale.all() > 0), pdb.set_trace()
>    1276             self.normal = np.vstack(
>    1277                 (normx, normy, np.zeros(len(scale)))) / scale
> 
> ipdb> 

File "lat_12_0095745_lon_57_695605.lay":

>  [info]
> format = cart
> version = 1.3
> type = outdoor
> 
> [points]
> -91795177 = (12.0095745, 57.695605)
> -91908745 = (12.0083382, 57.6957989)
> -145146022 = (12.0076976, 57.6959072)
> -145146024 = (12.0072021, 57.6960067)
> -145146026 = (12.0067792, 57.6961417)
> -145146031 = (12.0062519, 57.6964615)
> -145146033 = (12.0060802, 57.6966816)
> -145146034 = (12.0059833, 57.6967532)
> -145146035 = (12.0056081, 57.6969385)
> -145146037 = (12.0052734, 57.6970486)
> -145146039 = (12.0048958, 57.6970945)
> -145146041 = (12.0045009, 57.697099)
> -145146043 = (12.0041748, 57.697044)
> -158674217 = (12.009443, 57.6971785)
> -158674528 = (12.0091476, 57.6971758)
> -158674529 = (12.0087333, 57.697155)
> -158674530 = (12.0082721, 57.697101)
> -158674531 = (12.008049, 57.6970734)
> -158675349 = (12.0064299, 57.6960695)
> -158675350 = (12.0066063, 57.6957579)
> -158675351 = (12.0067179, 57.695524)
> -158675352 = (12.0067817, 57.6952575)
> -158675353 = (12.0068394, 57.6950309)
> -158675354 = (12.0068914, 57.6948503)
> -158675355 = (12.0070766, 57.6945143)
> -158675356 = (12.0072071, 57.6943267)
> -158675358 = (12.0073512, 57.6941237)
> -158675359 = (12.0077393, 57.6932625)
> -299245223 = (12.005592, 57.6958965)
> -299245224 = (12.0056317, 57.6958391)
> -299245226 = (12.0056555, 57.6957853)
> -299245229 = (12.0058962, 57.6952785)
> -299245857 = (12.0056706, 57.6956452)
> -299245858 = (12.0057852, 57.6954432)
> -299245859 = (12.0059054, 57.6951721)
> -299246103 = (12.0045848, 57.6955972)
> -299246104 = (12.0046517, 57.6956068)
> -299246105 = (12.0048996, 57.6956836)
> -299246106 = (12.0053254, 57.6958347)
[... 479 lines cut ...]
> 
> [files]
> furniture = 
> 
> [slabs]
> WALL = {'color': 'grey20', 'lmatname': ['BRICK'], 'lthick': [0.07], 'linewidth': 3}
> _AIR = {'color': 'white', 'lmatname': ['AIR'], 'lthick': [0.02], 'linewidth': 1}
> AIR = {'color': 'white', 'linewidth': 1, 'lthick': [0.1], 'lmatname': ['AIR']}
> CEIL = {'color': 'grey20', 'linewidth': 1, 'lthick': [0.1], 'lmatname': ['REINFORCED_CONCRETE']}
> FLOOR = {'color': 'grey40', 'linewidth': 1, 'lthick': [0.1], 'lmatname': ['REINFORCED_CONCRETE']}
> 
> [materials]
> BRICK = {'epr': (4.09999990463+0j), 'mur': (1+0j), 'sigma': 0.300000011921, 'roughness': 0.0}
> AIR = {'epr': (1+0j), 'mur': (1+0j), 'sigma': 0.0, 'roughness': 0.0}
> REINFORCED_CONCRETE = {'mur': (1+0j), 'epr': (8.69999980927+0j), 'roughness': 0.0, 'sigma': 3.0}
> 
> [outdoor]
> 
> [latlon]
> llcrnrlon = 12.0038535
> llcrnrlat = 57.6932625
> urcrnrlon = 12.0095745
> urcrnrlat = 57.6975853
> projection = cass
>

srisooksai, 2021-01-06:
I could solve this problem by changing the code in line#1612 
https://github.com/pylayers/pylayers/blob/d9bc71ffe26453f6b6b7a246d059b7c319c468b4/pylayers/gis/layout.py#L1612
from 
`bcart = cart`
 to
` cart == cart`.

closed 2021-01-08 by srisooksai

hikaru-coder, 2021-03-08:
Hi srisooksai,

I tried the above but it crashes my kernel. It does not work when change to bcart to cart.