pylayers/pylayers #392: 'Polygon' object has no attribute 'setvnodes_new'

isaac1369, 2023-05-08:
Hi,

I installed pylayers and I want to run the 4 lines Ray-Tracing simulation. But, I get an error:
====================
 File "C:\Users\dehmollaian\pylayers\pylayers\gis\layout.py", line 6173, in <listcomp>
    [ polygon.setvnodes_new(self.get_points(polygon), self) for polygon in lTP ]
AttributeError: 'Polygon' object has no attribute 'setvnodes_new'
====================
Can you help me to fix it?

Best,
Isaac

isaac1369, 2023-05-08:
Incorrect installation procedure!

isaac1369, 2023-05-08:
I fixed the issue. The issue was that the installed version of Shapely was not the one mentioned in the requirements.txt

summer2ice, 2025-05-09:
Hello, I also encountered the same problem. May I ask how you specifically solved it? version of Shapely? Thank you in advance

summer2ice, 2025-05-09:
@isaac1369  Hello, I also encountered the same problem. May I ask how you specifically solved it? version of Shapely? Thank you in advance